vsulako/AFFBWheel #1: stepper version???

gielbloks, 2022-04-03:
good day i am enjoying this info you share!

i have experience with steppers and power systems through
3d printing and cnc, this is the first sim rig i am building

i am building a sim wheel myself but want to use a double 3nm nema 23 beqause i have them lying around with open loop drivers,

for power i use 4 350 watt hp server psu's in series so 48 volt, otherwise the nema 23 are to slow
costing €20 euro per 10 pieces here

but can i use open loop system with an hall encoder you are using on the steering shaft? the steppers are going to be belt drive 1:4 so i have 24 nm at max, having a little reserve power is wished so the motors don't need to maxxed out and burn..

regards giel

vsulako, 2022-04-03:
Sorry, I think I do not understand your question well. 

If you want to use stepper in sim wheel controller as feedback motor - I doubt it can be done easily.
I actually tried that once with disappointing results.
Usually, you can control stepper rotating speed and direction. When there's no control signal, it's standing still. 
And it IS still - one almost cannot turn stepper's shaft by hand when it is powered. 
So, who needs a sim wheel which only turns by itself, but cannot be turned by hand?
I tried to enable/disable stepper at high frequency, and also lower stepper's current. It somehow helped, but caused stepper to jerk a lot. It was unusable and I did not experiment further.
Of course, you can experiment yourself, but I cannot help with that. 
At least, you will need to replace BTS7960 with stepper driver, and rewrite motor handling code to work with this driver.

"hall encoder" (I suppose, you mean TLE5010 or AS5600?) is merely a rotation angle sensor and have nothing to do with feedback motor.

gielbloks, 2022-04-03:
from the 3 d printing systems i always have the position the stepper is moved to 
then it stays powered to prevent the moving when an other stepper is running ( the z axis height) is always holding position beqause otherwise it could fall down..sort of standby current ist is adjustable from 10 to 80% typically using 25.. 

my thoughts where change the pwm signal into a step /dir signal but as i read it isn't that easy..
i also have closed loop drivers but they will always move to the commanded position so i think this couldn't be used eighther or do you think it will?

i was thinking stepper motors are more rigid than dc.. i have a big wiper motor and a starter motor from a truck.. but if you put force on it i was thinking it would burn out in a very short time or am i afraid for nothing?

gielbloks, 2022-04-03:
ill work it out thanks

vsulako, 2022-04-03:
of course dc motor can burn if it is powered and blocked (stalls). I suggest to use external current limiter, or power source that will not exceed stall current of motor.

gielbloks, 2022-04-03:
thinking about another thing,is it also possible to add 10 potentiometers more on it for brake balance etc etc,i am building a full f1 wheel made the wheel itself in carbon/glass fiber when ready i can pull a mold from it so i can duplicate them for 30/50€ 80/100€ including pedalbox and steer base with room for a screen on the base or the steer itself,ill include a picture from the frontplate ( remember it is in the rough!!!) 
[image]

vsulako, 2022-04-03:
You mean add 10 additional axes? No, there is a limit of 8 axes per generic controller.  
But simplest solution is to use additional controller. a ProMicro with MMJoy2, or STM32F103C8 with FreeJoy firmware will allow to add 8 axes/128buttons more. If game/sim can use such number of axes, it will defininely work with multiple controllers.

gielbloks, 2022-04-04:
thankyou!!