vsulako/AFFBWheel #2: Hand brake axis.

lazenyuk-dmitry, 2022-04-06:
I haven't used your firmware yet. But it looks cool. I'm still waiting for accessories from AliExpress.

But I think it would be cool to add a pin to connect the handbrake potentiometer.

vsulako, 2022-04-06:
It is possible to add 2 more axes. 
Will try to implement it later.

lazenyuk-dmitry, 2022-04-06:
> It is possible to add 2 more axes.

ok, that's what I'm planning to do now.

vsulako, 2022-04-06:
Added 2 analog axes similar to stick. No spare pins now :)

lazenyuk-dmitry, 2022-04-07:
А ты говоришь по русски. У меня с английским не очень. Напишу на руском.

Я думаю достаточно добавить только ручник. Я не понял у тебя совсем не осталось пинов свободных или не осталось свободных для осей? Вроде по схеме еще есть пины свободные. Потому что на свободный пин ты мог бы еще добавить подключение канала Z энкодера, как в этом проекте https://github.com/ranenbg/Arduino-FFB-wheel/issues/3. Можно искать автоматически центральное положение руля при подключении, по моему хорошая фича. Хотя ты может это планировал сделать. Правда энкодер с каналом Z стоит в 3 раза дороже.

Не знаю нужен ли thumbstick вообще. Можно конечно его вынести на руль для обзора в игре но можно так же это и кнопками в принципе делать. Возможно интереснее было бы использование этих осей в качестве КПП на осях, как здесь https://www.youtube.com/watch?v=aYYAbcPKN8g там в описании есть схема. Я слышал что есть КПП logitech которые именно так и работают на потенциометрах. Но опять же не знаю насколько это актуально и многим ли нужно. Ну а вообще тоже интересная фича. Или можно предоставить выбор использовать те оси как thumbstick или как shifter, например как настройку в программе или в прошивке.

Вот в общем добавить подключение Z канала энкодера и автоматический поиск центра. И можно сделать возможность подключения КПП на осях. И тогда в твоей прошивке неверно будет все)

vsulako, 2022-04-07:
Не осталось пинов ProMicro, которые оставались бы свободными во всех вариантах подключений. Дополнительные оси наверное не повредят (хотя увеличивается время опроса). Может быть, лучше сделать их отключаемыми. 
Использовать Z канал энкодера и автопоиск центра не планировалось - у меня нет такого энкодера, проверять не на чем. Да и не вижу никакой проблемы выровнять руль вручную и нажать кнопку центровки. 
Стик задумывался для обзора в ETS2. Это такие же оси как остальные, их можно использовать на что угодно.
Что до эмуляции КПП двумя осями... выглядит несколько сложно, в плане того как это все конфигурировать и настраивать. Тем более что непонятно, насколько это востребовано. Пока что этот проект использую только я, и то только в ETS2 :)

lazenyuk-dmitry, 2022-04-07:
> Пока что этот проект использую только я, и то только в ETS2 :)

Ну думаю будут пользоваться) Я обязательно попробую как получу arduino leonardo и прочие комплектующие.
Вообще не так много прошивок для arduino. А так вообще пины одни и те же что ProMicro что Leonardo?

Есть группа ВК с обсуждением самодельных рулей можешь заходить если интересно [Самодельный игровой руль. делимся опытом](https://vk.me/join/CbkLyOKf9HyRnVNvNuYgayWbtbIK9rxG7Yg=)

closed 2022-04-07 by lazenyuk-dmitry