vsulako/AFFBWheel #4: AS5600 Arduino Leonardo

mitiaborodada, 2022-04-22:
Приветствую! Никак не хочет работать датчик AS5600 с Arduino Leonardo. Подключаю пины датчика SDA и SCL к пинам ардуино - RXI TX0 соответственно.  В config.h прописываю:
#define I2C_PIN_SDA 0   
#define I2C_PIN_SCL 1

Но ничего не работает, никакой реакции. Причём сам датчик работает на этой же ардуинке с проверочным кодом.
Естественно, SDA и SCL подключаю к ардуино к соответствующим пинам SDA и SCL:

#include <AS5600.h>
AS5600 encoder;
int Wheel;
void setup() {
Serial.begin(115200);
}
void loop() {
Wheel = encoder.getPosition();
Serial.println(Wheel);
}

Уже голову сломал. Буду очень благодарен за помощь.

vsulako, 2022-04-22:
Используете последний код прошивки? Ранее была проблема изза которой AS5600 не работал.

mitiaborodada, 2022-04-23:
Да, скачивал буквально вчера. Может быть проблема из-за того, что кроме датчика AS5600 больше ничего не подключаю при этом? Сначала думал, что датчик сгорел, на этой же ардуинке с тестовым кодом работает и данные читаются.

vsulako, 2022-04-23:
Нет, должно работать и без всего остального. Попробуйте подключить к другим пинам. 
Строчки для энкодера комментировать не обязательно, они используются только если в  STEER_TYPE указан ST_ENCODER.

mitiaborodada, 2022-04-23:
Да, спасибо, всё получилось! Я исправил предыдущее свое сообщение. 
 Остался последний вопрос. 
Как мне использовать AS5600 на всех осях: руль + педали?

vsulako, 2022-04-23:
Датичк еще может не работать, если его подключить после старта (или в процессе работы у него пропадет питание), потому что при старте идет инициализация, а далее только опрос.
В этой прошивке поддержка AS5600 планировалась только на оси руля. На педалях можно разве что вместо потенциометров поставить датчики Холла. 
а чтобы AS5600 ставить на педали и другие оси, понадобится доработка. AS5600 еще и не позволяет менять I2C адрес, так что понадобятся ухищрения - что-то вроде TCA9548, либо на каждый датчик хотя бы по отдельной SDA линии, либо что-то еще - а это будет много изменений и в подключениях, и в коде. А у меня всего один датчик, я даже попробовать не могу :)

vsulako, 2022-04-23:
Вообще ведь у AS5600 есть еще аналоговый выход, так что дополнительные датчики можно подключать как потенциометры - выход OUT вместо среднего контакта потенциометра. Но я это не пробовал.

mitiaborodada, 2022-04-24:
Кстати, пробовал использовать 4 датчика с помощью TCA9548A. Отлично всё работает.

Всё таки, что-то не так и не пойму пока источник проблемы. Вчера всего лишь перепаял другой провод на датчике. Подключаю к ардуинке и снова ничего не работает - никаких сигналов! Тут же заливаю тестовую прошивку, переключаю SDA и SCL на соответствующие пины Arduino SDA и SCL и вуаля - всё работает! Переключаю обратно и ничего не работает. Пробовал с другими пинами, тоже ничего не работает.  
Ещё помучаюсь с этим датчиком и наверное буду переходить на обычный энкодер.

vsulako, 2022-04-24:
Есть предположение. Попробуйте в файле bb_i2c.cpp сверху изменить 4 строчки: 

```
#define I2C_SDA_LOW   {pinModeFast(I2C_PIN_SDA, OUTPUT);__builtin_avr_delay_cycles(4);}
#define I2C_SDA_HIGH  {pinModeFast(I2C_PIN_SDA, INPUT);__builtin_avr_delay_cycles(4);}
#define I2C_SCL_LOW   {pinModeFast(I2C_PIN_SCL, OUTPUT);__builtin_avr_delay_cycles(4);}
#define I2C_SCL_HIGH  {pinModeFast(I2C_PIN_SCL, INPUT);__builtin_avr_delay_cycles(4);}
``` 
и посмотреть не заработает ли.

mitiaborodada, 2022-04-26:
Да, всё получилось! Спасибо огромное! 
Вроде понял в чём была проблема. У меня датчики на 3,3 вольта. Я изначально подключал его к Arduino micro через преобразователь напряжения на 3.3 вольта. Всё работало без проблем, но на leonardo при подключении к пину 3.3 вольта и к RX TX работает не стабильно. Можно сказать, что вообще не работает. Даже с преобразователем не захотел работать. Но на свой страх и риск попробовал на 5 вольт подключить и о чудо, заработал на leonardo! Но лучше всё таки на 3.3. В итоге с вашими последними рекомендациями заработало на leonardo на 3.3 вольта, как и положено. 

И всё таки хочу попробовать помучиться подключить AS5600 через TCA9548A и КПП по двум проводам (на каждую кнопку у меня резистор добавлен), опять же, если получится.

Ещё раз, большое спасибо!

vsulako, 2022-04-26:
Код в последнем сообщении увеличивает задержки при обмене по I2C, я предложил его для диагностики. Если работает и без него, то лучше задержки (__builtin_avr_delay_cycles) убрать. 
Я был уверен что AS5600 (белый квадратный модуль) работает от 5в, так как контакт VCC соединен с первой ножкой чипа, и сразу его подключал к 5в. А оказывается, еще и со второй. Надо подумать что с этим делать. Вообще говоря, у меня он на леонардо и на 3.3в работает, но в руле promicro, там преобразователя 3.3 нет.
TCA9548A у меня нет в наличии, с ним помочь не могу. Для минимизации проблем можно повесить его на пины для аппаратного i2c и использовать готовые библиотеки. Код работы с осями педалей, по моему, должен быть вполне понятен, если что - спрашивайте.

vsulako, 2022-04-26:
Нашел - чтобы безопасно подключать модуль AS5600 к 5в, надо убрать резистор R1 (0ом).

vsulako, 2022-04-27:
Тут вот в чем дело. Я работаю с i2c устройствами через ручную манипуляцию пинами, так получается быстрее, и можно брать любые пины.
Но не все устройства могут работать достаточно быстро, поэтому приходится немного притормозить обмен, кое-где вставляя задержки. Я подобрал задержки под свои устройства, предполагая что все они одинаковые. Видимо, нет.
Предложенный код добавляет задержку в 4 такта на каждое действие с i2c. Попробуйте поуменьшать в нем значение для __builtin_avr_delay_cycles, от 4 до 1.

mitiaborodada, 2022-04-28:
Уменьшил задержку с 4 сразу до 1 - работает - так и оставлю! Видимо это особенность моих устройств, в первую очередь  моего экземпляра Arduino Leonardo, т.к. на микро всё работает по дефолту.

mitiaborodada, 2022-05-01:
Удалось реализовать педали на датчике AS5600 вместо резисторов. Удалил на плате датчиков 2 резистора: R1 (0 Ом) и R4

mitiaborodada, 2022-07-02:
Приветствую!
Я не знаю, что у меня происходит с датчиком AS5600, но он перестал работать на руле. (заказал себе TLE, посмотрю как он будет себя вести)
Суть в том, что AS5600 в один прекрасный день просто перестал реагировать на поворот магнита. Вообще никакие показания не считывает, как будто датчик мертв. Причём за день до этого было всё нормально. 
При этом, заливаю тестовый скетч и просто "перетыкиваю" коннекторы с датчика на SDA и SCL на соответствующие пины Leonardo - датчика работает. Всё четко считывает. Что за магия, понять не могу!  

Может есть какие-то идеи?

#include <AS5600.h>
AS5600 encoder;
int Wheel;
void setup() {
Serial.begin(115200);
}
void loop() {
Wheel = encoder.getPosition();
Serial.println(Wheel);
}

vsulako, 2022-07-02:
А задержки о которых тут речь выше, есть? которые __builtin_avr_delay_cycles
забыл их добавить, может, вы обновили прошивку?

если использовать "родные" i2c пины, то надо бы добавлять и "родной" i2c.
Примерно так: ищем код
```
#if STEER_TYPE == ST_AS5600
  #include "bb_i2c.h"
  #include "multiturn.h"
  
  AS5600_BBI2C AS5600;
  MultiTurn MT;

  #define SETUP_WHEEL_SENSOR setupAS5600();
  #define GET_WHEEL_POS (MT.setValue((AS5600.readAngle()-2048)<<(STEER_BITDEPTH-12)))
  #define CENTER_WHEEL MT.zero();

  void setupAS5600()
  {
     AS5600.begin();
  }
#endif
```
и меняем на 
```
#if STEER_TYPE == ST_AS5600
  #include <AS5600.h>
  #include "multiturn.h"
  AS5600 encoder;
  
  MultiTurn MT;

  #define SETUP_WHEEL_SENSOR
  #define GET_WHEEL_POS (MT.setValue((encoder.getPosition()-2048)<<(STEER_BITDEPTH-12)))
  #define CENTER_WHEEL MT.zero();

#endif
```
В работе не проверено, просто поменял соответственно вашему коду.
И еще придется освободить пины 2 и 3, так как они и есть "родные" SCL/SDA (те которые выведены отдельно на Leonardo - это они же), если на них что-то будет - будет мешать.

mitiaborodada, 2022-07-02:
Семён Семёныч, точно!  Я же обновился, а про эти настройки забыл. Причём после обновления датчик какое-то время работал, а потом просто "сдох". 
Спасибо тебе, добрый человек!

closed 2022-07-02 by mitiaborodada