vsulako/AFFBWheel #6: H-Shifter with 2 wires

mitiaborodada, 2022-04-28:
Пытаюсь разобраться в коде (если честно не силён в этом), чтобы понять как можно использовать h-shifter на 2 провода.
У меня на выходе шифтера , при переключении передач,  образуется сопротивление, которое зависит от номера передачи.
Я пробовал в тестовых программах, использующие библиотеку joystic, ипользовать мой h-shifter. Там очень легко можно реализовать под мои нужды. В вашем коде, пока не понимаю как это сделать. Пока только понял, что нужно сделать изменения в AFFBWheel.ino в секции  "reading buttons".

vsulako, 2022-04-28:
Для начала понадобится пин с аналоговым входом. На promicro лишних нет, но на Leonardo есть еще три незадействованных. Например, пусть это будет A4. 
Далее, пусть схема подключений условно такова:

[image]

Ваша может отличаться, но принцип одинаков.
При нажатии одной из кнопок напряжение на A4 меняется. Измерив его, можно узнать, какая из кнопок нажата.
Если нажато несколько кнопок, понять это будет сложно, но для КПП сойдет - там одномоментно нажимается только одна.
Если измерять это напряжение функцией analogRead(A4), получится примерно так:
1 кнопка - 720, 2 - 670, 3 - 600, 4 - 495, 5 - 330, ни одна кнопка не нажата - 1023. (вообще это можно просто взять и проверить отдельным скетчем)

теперь к коду.
Опрос кнопок проходит в функции readButtons.
Состояние кнопок хранится в виде битовой маски в переменной wheel.buttons, это 32-битное целое число.
Каждый бит означает одну кнопку, 1 - кнопка нажата, 0 - не нажата.
К концу выполнения функции состояние всех кнопок записывается в эту переменную.
У нас 5 кнопок, так что выделим под них 5 бит под номерами, скажем, 27-31.
Чтобы не заморачиваться, вставим наш код в самый конец функции и будем перезаписывать состояние кнопок новыми данными, вне зависимости от того что там прочиталось до этого. Антидребезг тоже будет игнорирован, ну да и ладно.

Так что, идем в самый конец функции readButtons, и добавляем чтение A4, выяснение какая кнопка нажата (добавим еще допуск плюс-минус 20) и установку соответствующих бит в wheel.buttons:

```
int val=analogRead(A4);
bitWrite(wheel.buttons, 27, ((val>700) && (val<740)));
bitWrite(wheel.buttons, 28, ((val>650) && (val<690))); 
bitWrite(wheel.buttons, 29, ((val>580) && (val<620)));
bitWrite(wheel.buttons, 30, ((val>475) && (val<515)));
bitWrite(wheel.buttons, 31, ((val>310) && (val<350)));

```
Навскидку как то так. Для другого количества кнопок все будет аналогично.

mitiaborodada, 2022-04-28:
Ну как готовый... Сам делал))) 6 передач + задняя. 
Спасибо за подсказку. Первая проба прошла почти удачно. Все кнопки работают, фиксируются, но при этом почему-то также срабатывают оси  Y и Z  При этом всё отключено - руль и педали.
[image]

vsulako, 2022-04-28:
Оси срабатывают, как раз потому что к ним ничего не подключено, Так они висят в воздухе и ловят каждый чих. Это нормально. Когда будут подключены, ничего такого не будет.

mitiaborodada, 2022-05-01:
Подключил все оси. Всё работает как надо! Спасибо за подсказку как сделать КПП на 2 провода.

closed 2022-05-01 by mitiaborodada

mitiaborodada, 2022-05-09:
Оказалось, что у меня был комплекс проблем и главная из них БП. Он просто давал чудовищные шумы! Из-за этого аналоговые входы давали непредсказуемые значения. Когда заменил БП, то всё встало на свои места. 

По данной схеме КПП у меня получились такие данные резисторов (может кому пригодится, кто будет делать такой же.): подтягивающий резистор - 10 кОм, резисторы на кнопках взял по 2.2 кОм. Работает идеально!