vsulako/AFFBWheel #7: Педали на датчиках AS5600

mitiaborodada, 2022-05-01:
На Arduino Leonardo с вашей помощью  удалось реализовать педали на датчиках AS5600.
При запуске программы AFFBWheelGUI всё работает как надо: руль, педали, кнопки.
Но, когда запускаю тот же ETS2, то в ней педали работают как кнопки - либо педаль включена, либо выключена.
Проверил и откалибровал в стандартной виндосовской настройке игровых устройств.  Тут тоже как и в  AFFBWheelGUI  всё настраивается и калибруется.
Но в игре не работает. Пробовал в игре BeamNG - та же история, что и в ETS.
В чём может быть проблема?

vsulako, 2022-05-02:
Очень странно. Если в свойствах контроллера Windows ось работает правильно, то и игры должны ее так видеть. Не знаю, в чем дело. 
С шумом осей от мотора тоже не сталкивался. Питание мотора отвязано от питания контроллера? У меня контроллер питается от USB, а мотор от отдельного БП.

mitiaborodada, 2022-05-03:
Вот и мне странно. Проект затянулся, всякий раз вылазит новый баг :)  Контроллер питается от USB, мотор от своего БП.
Ещё помучаюсь и проверю как будет рнаботать на Arduino micro.
Сейчас основная проблема - это шум педалей и при подключении КПП постоянный дрейф на аналоговом входе, даже когда не включена ни одна передача. Должно быть значение 1023, а оно постоянно дрейфует.

vsulako, 2022-05-03:
А когда передачи включены, шум есть? 
И если вместо AS5600 на педали подключить потенциометр, будет ли шум?

vsulako, 2022-05-04:
Если это только на аналоговом выходе, а цифровой в порядке - то можно его на ось руля применить.