vsulako/AFFBWheel #8: Обратная связь не работает (?)

mitiaborodada, 2022-05-01:
Приветствую! 
Почему-то в играх не работает обратная связь. Т.е. она ка бы есть, руль при крайних положениях останавливается - мотор его удерживает. При столкновениях и каких-то неровностях тоже дёргается, но нет возврата руля, когда на скорости делаешь поворот. Колёса должны возвращать руль в прямое положение, но этого не происходит. 
Есть идеи, почему так? Если бы обратная связь не работала, то всё остальное так же бы не работало.

vsulako, 2022-05-02:
Остановка руля на крайних положениях сделана независимо от команд обратной связи, игры ею не управляют.
Для проверки работы ОС можно использовать Fedit, ссылка есть в readme - там можно создать и запустить все доступные эффекты ОС.
В играх которые я пробовал, тоже не было этого эффекта возврата руля на повороте, как на реальном автомобиле. Сопротивление на поворотах было, тряска при столкновениях была, а возврата не было. Честно говоря, не знаю - то ли игры этого не умеют, то ли дело в чем-то другом.

lazenyuk-dmitry, 2022-05-03:
Я спросил в группе в ВК у ребят, они говорят что должен быть такой эффект как на реальном авто. Предполагают что проблема в прошивке. 

@mitiaborodada в какой игре у вас проблема такая проблема? И вы можете это проверить на других прошивках? Я пока не могу этого сделать не все детали есть. 
https://github.com/ranenbg/Arduino-FFB-wheel вот эта с открытым кодом, будет удобнее исправить если там все работает.
https://www.youtube.com/watch?v=aYYAbcPKN8g&t=6s а эта уже условно бесплатная, здесь в коде покопаться не получиться если что.

Если на этих прошивках работает все тогда проблема в прошивке. А возможно что то в настройках нужно покрутить.

mitiaborodada, 2022-05-03:
@lazenyuk-dmitry я проверял в ETS, BeamNG и TestDrive Unlimited, везде результат один и тот же. 
На других прошивках пока не могу проверить, т.к. у меня нет обычного энкодера, есть только датчики AS5600.  
https://github.com/ranenbg/Arduino-FFB-wheel  с этой можно попробовать, если удастся приспособить мои датчики.
Всё таки, думаю, что всё дело в прошивке. Прошивка vsulako меня вполне устраивает, но хотелось бы возврат руля после поворотов.  В любом случае спасибо @vsulako за открытый проект и что очень быстро отвечаете!

lazenyuk-dmitry, 2022-05-04:
Я тут немножко почитал по эффектам. 

За возврат руля в положение в котором стоят колеса должен отвечать эффект `SPRING` или `DAMPER`. Больше по описанию похоже что SPRING должен возвращать руль в нужноное положение. Нужно проверить работу этих эффектов наверно.

@mitiaborodada  попробуйте сделать следующие.
- через [AFFBWheelGUI](https://github.com/vsulako/AFFBWheelGUI), на вкладке `Force Feedback` сделайте больше `Min Force`, `All` возможно появится нужный эффект и в игре можно эти настройки проверить.
- само собой [протестите эффекты](https://github.com/vsulako/AFFBWheel#testing-software)
- ну и можно поиграться с ползунками в настройках.
Возможно что то получиться.

@vsulako есть еще настройка, например у Logitech:
[image]
`centering spring` - я так понял это в играх где плохо работает FFB или не реализован возврат руля. То есть даже когда машина стоит на месте и ты поворачиваешь руль то он стремиться в центральное положение вернуться. Обычно ее не включают. Но реализовать можно, может в каких то играх понадобится. 

https://support.thrustmaster.com/en/kb/107-en/#:~:text=A%20damper%20force%20controls%20how,on%20a%20spring%2Dmass%20system.
https://www.reddit.com/r/simracing/comments/9oqd7g/question_regarding_centering_spring/

vsulako, 2022-05-04:
Про эффекты.
Есть вот такой документ: https://www.usb.org/sites/default/files/pid1_01_0.pdf
Там вот как раз все об эффектах обратной связи, и как они должны описываться в HID-дескрипторе.
Таких эффектов придумали аж 12 штук.
Вот скажем CONSTANT - элементарно, игра просто говорит "нужно такое-то усилие в такую-то сторону" и все.
А вот RAMP - тут усилие должно в течение некоторого времени изменяться от одного значения к другому.
SINE - усилие изменяется по синусоиде с задаваемыми параметрами.
SPRING - усилие зависит от отклонения от некоторого условного центра, чем больше значение оси отклоняется от центра, тем больше возвращающее усилие, по линейной зависимости.
DAMPER - тут усилие зависит от разницы угловых скоростей, иначе говоря - чем больше скорость отклоняется от некоторой заданной, тем больше противодействующее усилие.
FRICTION - это должно ощущаться как трение, на практике - это усилие постоянной величины, всегда направленное против движения.
И игры должны давать команды устройству на управление этими эффектами, создавать их, удалять, изменять параметры.
В программе Fedit с этим прямо наглядно можно поиграться.

Ирония в том, что по сути, почти все эти эффекты не нужны.
Ну вот та же пружина. Ну хорошо, этот эффект возвращает ось в центр. А толку-то? В реальности все действует совсем не так.
Вот автомобиль стоит. Выворачиваем руль в сторону - он не возвращается. Теперь плавно нажимаем на газ - и только тогда колеса начинают испытывать сопротивление, выравнивающее их, что и передается на руль, когда он возвращается в центр. Такое усилие зависит от скорости движения, от свойств дороги (на льду, например, будет совсем не так как на асфальте, и если яма попадется - должно дернуть), и может еще от чего . На простой эффект возвращающей пружины это никак не походит. С остальными эффектами то же самое - они все синтетические.
Больше всего знает от том, что должно происходить с обратной связью, какое усилие должно возникать на руле - только сама игра. Только там можно все это просчитать правильно, учитывая все параметры.
Вот и выходит, что из всего набора эффектов игре по сути нужен только самый простой - CONSTANT. Чтобы игра могла получить данные о положении руля, и обратно сообщить, какое усилие создать. И все это как можно чаще, этак раз в миллисекунду. Быстродействия компьютера на это вполне хватит. И так можно проэмулировать вообще все эти эффекты.
BeamNG, к примеру, ровно так и делает. Это даже у них в справке написано. Так делает ETS2 последних версий - ранее там все еще использовались эффекты пружины и демпфера, а теперь - только CONSTANT.
Я это смотрел выводя все поступающие команды.

И потому совершенно непонятно, почему, например, в современном ETS2 нет этого эффекта возврата руля. Если хоть какие-то эффекты есть - должно работать все, потому что оно все делается через одну и ту же команду создания эффекта CONSTANT. Такое впечатление что игра этого не умеет.
Кстати, не помню, было ли так в BeamNG, там как мне показалось, самая продвинутая модель. надо попробовать.

насчет centering spring - это в драйверах заводских рулей бывает такая опция. Что можно включить дополнительный эффект возвращающей в центр пружины, которым игры управлять не будут. Обычно это советуют выключать, потому что этим должна командовать сама игра. А вообще оно существует для игр без обратной связи, чтобы получить эффект как на руле без ОС, с обычной пружиной для возврата.
В стандарте HID/PID ничего такого нет, и на совсем старые игры я и не рассчитывал, так что такого добавлять не стал.

Кстати, готового руля с ОС у меня никогда не было, я потому и начал делать свой :) Потому нет образца для подражания.
Я пробовал пару доступных тогда других прошивок - RFR Ai-Wave и EMC, и насколько помню, в ETS они действовали довольно похоже.

lazenyuk-dmitry, 2022-05-04:
У меня тоже нет руля с ОС) Сегодня пришли основные запчасти. Буду собирать)

RockKenny, 2022-05-04:
Проверил BeamNG на Logitech momo. Эффект выравнивания руля при движении есть. Если я правильно понял задачу. На стопе повёрнутый руль не выравнивается.

vsulako, 2022-05-04:
Да, сейчас опробовал - в BeamNG работает. На скорости руль сам старается выровняться, чем больше скорость - тем сильнее.

mitiaborodada, 2022-05-04:
> Да, сейчас опробовал - в BeamNG работает. На скорости руль сам старается выровняться, чем больше скорость - тем сильнее.

Т.е. на вашей прошивке возврат в центр на скорости работает? Видимо я в игре достаточную скорость не набирал. Всё пока тестирую на не закрепленных КПП, руле и педалях (((

vsulako, 2022-05-04:
В BeamNG - определенно. Хотя я там в настройках задрал для нее уровень.
[image]
А вот в ETS2 что-такое есть если выкрутить на максимум настройку "возврат руля на низкой скорости", будто бы пытается, но никак не может. Там еще есть "возврат руля на высокой скорости", и эта как будто даже не работает. Есть сопротивление на поворотах, но возврата не происходит.

mitiaborodada, 2022-05-04:
Ага, понятно! Настройки беру на заметку. Как только всё закреплю, тоже видео выложу. Есть что показать в плане механики)))  Много чего хочется реализовать, например, механизм автоматического отщёлкивания поворотников. Можно конечно готовый механизм использовать от каких-нибудь Жигулей, но это не наш метод) Но это уже другая тема.

mitiaborodada, 2022-05-30:
У меня ещё такой вопрос по обратной связи. В ETS2 и ATS, когда у грузовика заглушешь двигатель, то руль начинает дёргаться мелкой дрожью, даже. Это присходит, даже если после этого выйти в меню. Как только снова запускаешь движок, дрожь прекращается. Почему так, с чем может быть связано? Это только у меня?

lazenyuk-dmitry, 2022-05-30:
У меня так же в city car driving вроде. Попробуй в программе для настройки убавить эффект friction, он второй снизу. Я уменьшал процентов на 20 можно вообще вырубить попробовать. На emc прошивке так же у меня)

vsulako, 2022-05-30:
У меня вроде такого нет.

mitiaborodada, 2022-05-31:
А вот у меня вопрос-идея. Можно ли сделать искусственную "пружину" для тех игр, где нет возврата руля в центр, как например, в ETS2? Допустим по нажатию кнопки во время игры включать этот эффект.

lazenyuk-dmitry, 2022-05-31:
> А вот у меня вопрос-идея. Можно ли сделать искусственную "пружину" для тех игр, где нет возврата руля в центр, как например, в ETS2? Допустим по нажатию кнопки во время игры включать / отключать этот эффект.

А вот с этим фиксом можете проверить? https://github.com/vsulako/AFFBWheel/commit/c49b4807198ac339c320e7df1ce2ae53413ceba4

Возможно будет работать этот эффект.

Пружину в принцине можно сделать.. Но я думаю это не так прикольно будет.