vsulako/AFFBWheel #10: Подключение драйвера мотора

lazenyuk-dmitry, 2022-05-28:
Привет @vsulako, заполни что ли профиль на гите а то не знаем даже как тебя звать)

В общем потестил я еще прошивку EMC и был немного удивлен, руль стал более тугим, пропала пустота в центре, и в целом отдача стала лучше и мощнее в раза так 2-3. Стал лучше эффект возврата руля в центр, да и вообще руль более нагруженным стал, стало заметно тяжелее крутить руль когда машина стоит на месте. В Dirt 2 по гравию у меня на твоей прошивке руль работает как перфоратор) На EMC лучше стало. И вот решил я попробовать немного по экспериментировать со схемой подключения на твоей прошивке. И так подключаю я `-5 v` на контакты `L_EN` и `R_EN`, вместо подтяжки, и стал фидбек мощнее и приятнее как то как на EMC прям. Еще можно просто перекинуть резистор подтяжки с `GND` на `VCC` на драйвере мотора, если захочешь проверить, но мне так нравиться отдача больше.  Но есть нюанс, на твоей прошивки при таком подключении руль постоянно тяжелый, нагружен, а на EMC на рабочем столе или в настройках игры он легкий. Ниже расскажу почему так, ну я предполагаю) Но это не точно)

Вот как это выглядит на схеме EMC:
[image]

На схеме EMC есть провод от контакта `8` до контактов `L_EN R_EN` я думаю что в нем все дело. Я померял тестером и на контакте  `8` появляется +5V именно тогда когда должны включаться моторы, наверно поэтому руль свободный вне игры. 
Ты можешь добавить такой же контакт в прошивку? Вроде еще есть свободные пины на плате? Просто попробуй выдавать +5V на L_EN и R_EN именно когда должны включаться моторы. Можешь это сделать в отдельной ветке? Я потестирую, да и ты тоже можешь, а потом смержишь в мастер ветку. Ну и попробуй так реально отдача мощнее и приятнее.

И вот еще момент... Есть эффект `Endstop` он отвечает за ограничение угла руля? Если я правильно понял) Так вот мне не хватает этого эффекта и он ведет как то себя странно. Я ставлю его до 200 процентов, например 150 и тогда становится чуть мощнее, но если ставлю выше, например 300-400 то тогда есть небольшой упор и потом руль прокручивается без усилия. Так вот можно ли как то этот эффект сделать мощнее? Я хотел бы более сильного ограничения в упоре. 

У меня если что два мотора 775 остальное ка на схеме драйвер мотора  BTS  и Arduino Leonardo.

lazenyuk-dmitry, 2022-05-28:
Если есть вопросы пиши или в ВК.

lazenyuk-dmitry, 2022-05-29:
Я сделал вывод 5V на 5 пин когда включается отдача и стало работать как на EMC один в один. Эффект возврата руля в центр стал прям намного сильнее и теперь я думаю в  beamng drive не придется повышать настройки фитбека чтобы все работало. Может на неделе попробую сделать получше и сделать пул реквест.

vsulako, 2022-05-30:
> на контакте 8 появляется +5V именно тогда когда должны включаться моторы
Так у меня же это сделано электрически. Два диода и резистор рядом с BTS7960. Когда на одном из каналов PWM есть +5 - на L_EN/R_EN тоже присутствует +5, когда на обоих 0 - на EN тоже 0 и питание с мотора снимается. Как раз против постоянной "нагруженности", странно что у вас она есть. 
Я пробовал управление через отдельный пин, но не понравилось то что руль становится "нагруженным" при любом значении ОС отличном от 0. Впрочем если надо то добавить несложно.

Эффект Endstop посмотрю.

> два мотора 775
интересно, как это механически сделано? оба мотора крутят один вал? Не получается ли так что один из них всегда халявит?

vsulako, 2022-05-30:
Добавил возможность задать пин для управления мотором (+5в когда усилие есть, 0 когда нет) в config.h (раскомментировать `#define MOTOR_ENABLE_PIN      5`) и поправил работу усиления эффектов.

lazenyuk-dmitry, 2022-05-31:
> поправил работу усиления эффектов

Спасибо стало лучше) Прокручиваться руль перестал в упоре.

lazenyuk-dmitry, 2022-05-31:
> Так у меня же это сделано электрически. Два диода и резистор рядом с BTS7960. Когда на одном из каналов PWM есть +5 - на L_EN/R_EN тоже присутствует +5, когда на обоих 0 - на EN тоже 0 и питание с мотора снимается. Как раз против постоянной "нагруженности", странно что у вас она есть.

Я это понял схему с диодами) Но когда напрямую подается питание на EN контакты тогда правда фидбек мощнее.
Если у вас как то получиться иначе решить это то было бы интересно. Наверно через диоды с контакта 9 и 10 не хватает питания на EN контакты.

lazenyuk-dmitry, 2022-05-31:
> Я пробовал управление через отдельный пин, но не понравилось то что руль становится "нагруженным" при любом значении ОС отличном от 0. Впрочем если надо то добавить несложно.

Так может так и должно быть?) Когда на месте машина стоит и крутишь руль он должен туго крутиться.

lazenyuk-dmitry, 2022-05-31:
> интересно, как это механически сделано? оба мотора крутят один вал? Не получается ли так что один из них всегда халявит?

Да тут все просто) Я так вот сделал.
[image]
[image]

Вроде не филонят моторы. Проверял по отдельности моторы в половину где то меньше по одному давят.

vsulako, 2022-05-31:
Ну это игра должна решать, насколько тяжело или легко должен крутиться руль. Рулю неизвестно, грузовик там в игре или велосипед, есть усилитель или нет :)
Я подумал что торможение при подаче питания на руль получается паразитное, и хорошо бы от него избавиться.
Ну вот при 0% все понятно - мотор надо выключить. При 100% - включить на всю. А если скажем игра говорит что нужно усилие 1% направо, она чего хочет? Наверное, очень слабое усилие вправо по сравнению с 0%. 
Но из-за особенностей реализации получается что в нагрузку с 1% вращения направо остальные 99% периода ШИМ происходит торможение, которого нет при 0%. Должно ли быть так? Не знаю.

Насчет падения на диодах - не думаю, по даташиту BTS7960 у него цифровые входы, все что больше примерно 1.75в считается за 1.

lazenyuk-dmitry, 2022-06-01:
> Насчет падения на диодах - не думаю, по даташиту BTS7960 у него цифровые входы, все что больше примерно 1.75в считается за 1.

Я думал что там падение идёт. Я думал что там маловато напряжения. 

А получается 5 пин это ШИМ? А можно ли как то не тупо 0 или 5В подорвать, а ШИМ сигнал?

vsulako, 2022-06-01:
>как то не тупо 0 или 5В подавать, а ШИМ сигнал

В теории можно (кроме случая когда применяется TLE5010, там на 5м пине выход тактового сигнала), только зачем? Попробовать сделать настраиваемую "тяжесть" руля?

lazenyuk-dmitry, 2022-06-01:
> В теории можно (кроме случая когда применяется TLE5010, там на 5м пине выход тактового сигнала), только зачем? Попробовать сделать настраиваемую "тяжесть" руля?

Да нет. Просто вы писали выше про паразитное торможение руля. Вот я и подумал может как то можно сделать интереснее а не подавать 0 или 5В. 

Типа так как сейчас вы сделали на 5 пин подачу сигнала если есть усилие. Так вот возможно можно регулировать это усилие с помощью ШИМ сигнала? И прилагать усилие которое хочет игра.

vsulako, 2022-06-01:
Ну да, можно, это же еще один способ подключения BTS7960, когда есть только один канал ШИМ.
Тогда ШИМ подается на оба EN, а вот на L_PWM/R_PWM - напротив, подаются 0 или 5в без ШИМ в зависимости от требуемого направления. В этом случае мотор будет отпускаться в холостой части периода. Я так не пробовал, но по моему, будет то же самое что и в случае с диодами.

vsulako, 2022-06-01:
ну вот [измененный файл](https://github.com/vsulako/AFFBWheel/blob/altmotorpwm/AFFBWheel/motor.cpp), можно пробовать.  пин 5 на EN,  9 и 10 - на PWM.

lazenyuk-dmitry, 2022-06-01:
> ну вот [измененный файл](https://github.com/vsulako/AFFBWheel/blob/altmotorpwm/AFFBWheel/motor.cpp), можно пробовать. пин 5 на EN, 9 и 10 - на PWM.

Да все так как вы и говорили, работает как на диодах. Мне больше нравится вариант с питанием по 5 пину. И кажется это более правильный вариант. В Assetto Corsa так лучше чувствуется авто и легче управлять. А ещё если выключить ABS то когда давишь по тормозам на скорости, то руль становиться лёгким.

closed 2022-06-01 by lazenyuk-dmitry

lazenyuk-dmitry, 2022-06-01:
А вот ещё момент небольшой. Когда в край выкручивает руль до упора с усилием, а endstop эффект стоит на 100%, то вроде все хорошо стопорит прям. А вот если выкручивает до упора с маленькими усилием, то руль как бы упирается а потом отскакивает назад и снова упирается и отскакивает, и снова, и это похоже на перфоратор) дребезжит в крайнем положении. Можно ли как то это сделать получше? Чтобы не было такого дребезга а держало руль.

lazenyuk-dmitry, 2022-06-02:
> Есть предположение.
> в основном файле, в строке 295 попробуйте поменять

Вроде помогло) Вы это добавите в мастер ветку?