vsulako/AFFBWheel #11: Программное отключение поворотников, используя обычные кнопки.

mitiaborodada, 2022-05-31:
Я тут сделал механизм отщёлкивания поворотников, поставил 2 кнопки, но они естественно работают не так как нужно, пришлось придумывать алгоритм. В этом деле не силён, но главное работает и алгоритм  себя отлично показал в игре. 

Суть его проста. Когда включается поворотник, замыкается соответствующая кнопка. Когда при возврате руля кнопка отщёлкивается нужно, чтобы в игре поворотник выключился. Програмно реализовал повторное виртуальное кратковременное включение кнопки, чтобы отключить сигнал поворота. Автор, может моя идея понравится и реализуете данную функцию лучшим образом, чем я. 

В начале файла AFFBWheel.ino поставил глобальные переменные:  
bool wr = false; 
bool wl = false;
bool flag = false;
int my_timer = 0;
bool Flag_RightSwitch = false; 
bool Flag_LeftSwinch = false;

В файле AFFBWheel.ino в конце секции readButtons() прописал такой код:

bool RR = digitalRead(4)^1; //если включен правый поворот, то состояние = 1
bool RL = digitalRead(7)^1; //если включен левый поворот, то состояние = 1

if ((RR) && (flag==0) && (Flag_RightSwitch==0)) {
  flag = true;
  Flag_RightSwitch = true; 
  my_timer = 0;
  wr = 1;
} else 
 if ((RL) && (flag==0) && (Flag_LeftSwinch==0)) {
 
   flag = true;
   Flag_LeftSwinch = true; 
   my_timer = 0;
   wl = 1;
 } 

 if(Flag_RightSwitch){
  if ((flag)&&(RR!=1)){    
    if(my_timer < 200){      
      wr = 0;
    } else wr = 1;
  my_timer++;
  }
 
  if(my_timer >= 400){
    wr = 0;
    my_timer = 0;
    flag = false;
    Flag_RightSwitch = false; 
  } 
 } else 

 if(Flag_LeftSwinch){
  if ((flag)&&(RL!=1)){
    
    if(my_timer < 200){
      my_timer++;
      wl = 0;
    } else wl = 1;
  my_timer++;
  }
 
  if(my_timer >= 400){
    wl = 0;
    my_timer = 0;
    flag = false;
    Flag_LeftSwinch = false; 
  } 
 }

bitWrite(wheel.buttons, 10, wr);
bitWrite(wheel.buttons, 11, wl);

lazenyuk-dmitry, 2022-05-31:
@mitiaborodada, вы можете сделать пул реквест. Тогда автору будет проще понять что вы сделали и где что написали. Так же он может там прокомментировать ваш код или улучшить, или даже принять ваши изменения. Это будет удобнее)

https://git-scm.com/book/ru/v2/GitHub-%D0%92%D0%BD%D0%B5%D1%81%D0%B5%D0%BD%D0%B8%D0%B5-%D1%81%D0%BE%D0%B1%D1%81%D1%82%D0%B2%D0%B5%D0%BD%D0%BD%D0%BE%D0%B3%D0%BE-%D0%B2%D0%BA%D0%BB%D0%B0%D0%B4%D0%B0-%D0%B2-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D1%8B

Информацию найти не сложно. Да и делается это тоже просто все.

vsulako, 2022-05-31:
Насколько понимаю, в механизме у вас кнопки поворота нажаты, когда рычаг поворотника находится в крайнем положении, когда возвращается в среднее - кнопки отпускаются.
А в играх поворотники включаются и выключаются кратким нажатием кнопки.
Так что вы имитируете короткое нажатие кнопки руля всякий раз как реальная кнопка нажимается или отпускается, так?

По моему, подобная функция есть в Joytokey и ей подобных программах.

mitiaborodada, 2022-06-01:
> Насколько понимаю, в механизме у вас кнопки поворота нажаты, когда рычаг поворотника находится в крайнем положении, когда возвращается в среднее - кнопки отпускаются. А в играх поворотники [...]

Да, совершенно верно - кнопка отпускается, после кратковременной паузы происходит короткое нажатие кнопки. Не стал использовать таймер с прерыванием (да и не силён в этом), а использовал обычный счётчик. Главное, что работает. Главное в настройках игры убрать галку автоматическое отключение поворотников.