vsulako/AFFBWheel #12: Потенциометры и группа для обсуждений

lazenyuk-dmitry, 2022-06-10:
Привет. У меня тут два вопроса есть.

1. Что можно использовать вместо дешевых потенциометров для педалей? Какие то магнитные датчики может? И может что то даже что было бы удобнее монтировать в педали. Или может какие то хорошие потенциометры можешь рекомендовать?

2. Ты не думал завести группу для обсуждений и тд? Думаю было бы нормально. У кого то может есть вопросы или сложности или какие то идеи интересные и не только по прошивке. И было бы еще интересно таким способом получать уведомления о каких то фиксах и новых фичах в прошивке. А то я смотрю ты много чего исправил, а я и не знаю) Мне лично нравиться то как группы в "телеграмм" работают, там и закреплять сообщения можно, что удобно было бы для каких то оповещений и тд. Ну и не создавать issues по мелким вопросам.

vsulako, 2022-06-10:
1. ну даже не знаю. У меня в педалях самые обычные потенциометры WH148 (да и сами педали от б\у руля, пробовал делать полностью свои - получилось неудобно). Понятно что скользящий контакт стирается и со временем будут работать хуже, но поменять их легко и стоят дешево. И монтировать удобно (хотя много зависит от конструкции педали)  - сверлим отверстие, вкручиваем, связываем с педалью через тягу или зубчатую передачу, все. Хотя все равно сложно рассчитать так чтобы использовать все 270 градусов поворота. Как то пробовал датчики Холла (OH49, SS495) - ничего, работают, тоже дешевые, потенциально вечные, подключаются прямо вместо потенциометров, но там больше требований к монтажу (отдельно магнит на оси закрепить, отдельно датчик, да чтобы зазор был одинаковый, да чтобы рядом ничего магнитного...) Вроде есть датчики Холла в виде крутилки, но чего-то очень дорогие. Еще тензодатчики есть (loadcell), могут замерять давление. Есть несколько штук, поигрался с ними, в планах добавить поддержку. Одни плоские - эти, похоже, можно прямо на педаль прикрутить под накладку и готово. Другие в виде бруска, работают на изгиб - но тут не знаю, как их в педаль вкрутить. 
2. а вот прямо тут чего то такое есть: https://github.com/vsulako/AFFBWheel/discussions

closed 2022-06-10 by lazenyuk-dmitry

lazenyuk-dmitry, 2022-06-10:
> 1. Еще тензодатчики есть (loadcell), могут замерять давление

Ну это на педаль тормоза актуально. И да можно добавить поддержку.

lazenyuk-dmitry, 2022-06-10:
> 1. да и сами педали от б\у руля, пробовал делать полностью свои - получилось неудобно)

Как сделаю свои попробую поделюсь чертежами если что)

mitiaborodada, 2022-06-10:
> Привет. У меня тут два вопроса есть.
>
> 1. Что можно использовать вместо дешевых потенциометров для педалей? Какие то магнитные датчики может? И может что то даже что было бы удобнее монтировать в [...]

Я на педали поставил датчик AS5600 в режиме аналогового варианта, подключается как  переменный резистор. С корпусом не стал заморачиваться и использовал обычные крышки от пластиковых бутылок - датчики как раз отлично там сидят
[image]
[image]
.

lazenyuk-dmitry, 2022-07-18:
@vsulako, Все же думаю как бы сделать педали так чтобы они более плавно передавали сигнал. У меня педали на потенциометрах WH148. Так вот все же сложно задействовать весь оборот потенциометра, поэтому как то немного рвано выглядят показания с потенциометра. AS5600 крепить как то сложновато. Вот есть были бы решения типа ужу готовые на AS5600. Типа потенциометр но магнитный. И вообще может можно сопротивление потенциометра попробовать меньше или больше, чтобы более плавный сигнал получить?

lazenyuk-dmitry, 2022-07-18:
> 1. Как то пробовал датчики Холла (OH49, SS495)

А как вы их подключали и как можно установить на педали?

lazenyuk-dmitry, 2022-07-19:
@vsulako, спасибо. Все вроде понятно и без рисунков. Попробую заказать потенциометры на 180 градусов и SS49E. Поэкспериментировать.