vsulako/AFFBWheel #14: ??

Brunno93, 2022-06-27:
wich version of delphi may i use to open GUI without errors? or, maybe you can create an exe program for calibrations

vsulako, 2022-06-27:
I used delphi 10.4.
complied exe are here: https://github.com/vsulako/AFFBWheelGUI/releases

Brunno93, 2022-06-28:
Many thanks, now i can adjust as i wish. i was using arduino IDE to adjust.
so, is it possible to add Multiplexers (ex: 4051) for buttons instead of HC165 (its difficult to find in my country) and a way to active, configure H_shifter on GUI.

vsulako, 2022-06-28:
With 4051 at least 7 pins are required to read 32 buttons, it seems too much. 
What about ordering online? HC165 and MCP23017 are available worldwide on Ebay or Aliexpress.
also I think of adding support for i2c expanders PCF857x.
configuring h-shifter in GUI - you mean analog h-shifter? maybe sometime later.

Brunno93, 2022-06-28:
4051 was for example only, but there is 4067 that have 16 in/output pins.
yes, analog H-Shifter, its looks perfect for me in particular.
I'm trying to understand each peace of code, i was thinking in a way that the program auto detect if some components are attached or not, for example, Analog H-shifter, pedals (accelerator, brake and clutch)

vsulako, 2022-06-28:
Still, 4067 need 4 pins for addressing and 1 for input, that takes at least 6 pins for 32 buttons. My setup (with Promicro) even does not have so much free pins.
Sure, it can be done, but for me, shiftin registers and i2c expanders look way more convenient for buttons input.

I think autodetecting is not possible. Components are too simple to tell something like 'i'm here'. Analog input will measure voltage, no matter if it has potentiometer connected or not. Only difference - if nothing is connected, it will receive noise which is not reliably distinguishable from useful signal.