vsulako/AFFBWheel #15: HC165

lazenyuk-dmitry, 2022-06-29:
Привет. Я тут подключил кнопки пока только по 1линии (DATA1)

И они работают но не правильно. У меня постоянно нажата кнопка 9 а вместо кнопки 13(вроде бы 13) нажимается кнопка 1.
И перепутаны кнопки вместо кнопки 1 нажимается 2 и тд 2 это 3, 3-4, 4-5....

Я использую  `SN74HC165N`. 
[image]

В чем может быть дело. Вроде собрано все правильно. Или у меня неправильный какой то HC165 но я нашел только такой.
Не охото все разматывать) Но вроде все правильно. Как будто смещены кнопки.

vsulako, 2022-06-29:
ой, похоже закралась ошибка. Обновил код, попробуйте теперь.

lazenyuk-dmitry, 2022-06-29:
Вот здесь https://github.com/vsulako/AFFBWheel/commit/4360836bd9587b872d9a16cf1384bb67b19e3e8c инпут лаг у кнопок начинается, в этом коммите.

vsulako, 2022-06-29:
странно. я лага не наблюдаю. и коммит этот никак с кнопками не связан...
а если загрузить настройки по умолчанию?

lazenyuk-dmitry, 2022-06-30:
У меня похоже как то выставился Debounce у кнопок, я не выставлял. Теперь все нормально) Спасибо.

closed 2022-06-30 by lazenyuk-dmitry