vsulako/AFFBWheel #21: Сдвиговые регистры.

Mozasvit, 2022-08-16:
Здравствуйте. Это снова я - неэлектронщик.

В инструкции написано брать 4 сдвиговых регистра 74hc165. На AliExpress продаются всякие разные. Взял sn74hc165n. Они подходят, или надо другие искать?

lazenyuk-dmitry, 2022-08-16:
@Mozasvit, у меня такие же. Все работает отлично. 

[image]

Mozasvit, 2022-08-16:
Я собрал частичный макет схемы, но руль выдает какую то ахинею.

lazenyuk-dmitry, 2022-08-16:
Если интересно, я распаявал сразу на минус перемычку и сопротивления 10к сразу на регистре а потом припаявал провода. Так удобнее. Фото жаль нету.

lazenyuk-dmitry, 2022-08-16:
> Я собрал частичный макет схемы, но руль выдает какую то ахинею.

Какую?

Mozasvit, 2022-08-16:
Фото будет позже. Нет доступа.

Показания энкодера руля скачут как бешеные. Допускаю неправильное подключение его в схему, там немного иная цветовая маркировка.

Показания осей педалей скачут. Возможно из-за шума. Я подцепил для пробы только один потенциометр.

Но GUI приложение из описания говорит мне, что кнопки 25, 26 нажаты, хотя кнопки не подключались вообще.

vsulako, 2022-08-16:
> В инструкции написано брать 4 сдвиговых регистра 74hc165. На AliExpress продаются всякие разные. Взял sn74hc165n. Они подходят, или надо другие искать?

Если в названии есть 74hc165 - это все одно и то же: https://www.chipfind.ru/crossreference/search.htm?s=74hc165 (разница в производителе и в несущественных мелочах)
Но для неэлектронщика, наверное, было бы проще использовать I2c расширители - меньше соединений и меньше возможности что-то напутать.

Mozasvit, 2022-08-16:
С учётом времени на доставку китайцами, этот вариант я опробую не скоро.

vsulako, 2022-08-16:
С энкодером: попробуйте добавить подтяжки на линии A и B энкодера. (т.е. резисторы 1-10кОм, от A/B к +5в)
С осями: неподключенные оси будут ловить шум, а что с подключенной?
С кнопками: вы случайно не включили ли функцию аналоговых кнопок? (не раскомментировано ли `#define APB`) В файле конфигурации для примера как раз стоят кнопки 25 и 26.

Mozasvit, 2022-08-16:
Аналоговы кнопки в настройках вроде не  трогал.

Позже проверю все методы. Сейчас нет доступа к плате.

Потенциометр, подключенный к пину педали газа, не реагирует на прокручивания.

Mozasvit, 2022-08-16:
Ещё нюанс.
Я как истинный неэлектронщик развел плату для этого проекта.
Это могло сказаться на конечном результате

lazenyuk-dmitry, 2022-08-16:
> Ещё нюанс.
> Я как истинный неэлектронщик развел плату для этого проекта.
> Это могло сказаться на конечном результате?

Что значит "развел плату"?

lazenyuk-dmitry, 2022-08-16:
> Показания осей педалей скачут. Возможно из-за шума. Я подцепил для пробы только один потенциометр.

В GUI можно пока отключить не используемые оси. Там галочка ставиться. Чтобы не мешало.

Mozasvit, 2022-08-16:
Я - "нарисовал", китайцы - "распечатали".

lazenyuk-dmitry, 2022-08-16:
> Потенциометр, подключенный к пину педали газа, не реагирует на прокручивания.

Странно. Если можете покажите как подключили.

Mozasvit, 2022-08-16:
Все посмотрю когда прийду домой.
Спасибо.

closed 2022-08-18 by Mozasvit