vsulako/AFFBWheel #25: Что то не понятное с кнопками.

lazenyuk-dmitry, 2022-09-11:
@vsulako, привет. Я тут сделал КПП это кнопки с 17 по 32. Теперь при подключении КПП перестают работать кнопки на руле 9-16, работает только 16 кнопка а 15 кнопка тригерит кнопку 9... Вобще какая то дичь. После отключения КПП все нормально. 
На на КПП у меня кнопки 17,19 и 18,20 это преключатели ВКЛ-ВКЛ. Теперь когда нажимаю 29 кнопку то все остальные отключаются, а при нажатии  кнопки 28 включается 29, 31, 32. Не понятно что то. Проблема с кнопками с 17 по 32. Что то с прошивкой или у меня?

vsulako, 2022-09-11:
Кнопки на регистрах HC165? Проверил у себя, ничего такого нет. Наверное, что-то в соединениях.

lazenyuk-dmitry, 2022-09-11:
> Кнопки на регистрах HC165? Проверил у себя, ничего такого нет. Наверное, что-то в соединениях.
> 

Да на регистрах HC165... Странно. Вроде как то проверял было нормально.... Вроде бы.

lazenyuk-dmitry, 2022-09-12:
@vsulako я нашел баг. Смотри регистры работают по парно можно сказать. Баг проявляется если подключить все кнопки, то есть все 4 регистра. Проблема проявляется при постоянном зажатии одной, двух кнопок. 

1. Подключите все 4 регистра все 32 кнопки.
2. На одном из регистров зажмите одну или две кнопки на постоянно. 
3. На втором парном регистре понажимайте кнопки, при нажатии некоторых кнопок проявляется баг. То есть если зажали на постоянно кнопки 17-24 то при нажатии некоторых кнопок 25-32 будут глючить кнопки. 

Если подключить только кнопки 1-16 то этот баг я не видел. Проверьте пожалуйста. У меня вроде последняя версия прошивки и GUI.

vsulako, 2022-09-12:
У меня не повторяется. Зажимаю 17-18, все 25-32 работают как ожидается, хоть по одной, хоть все вместе.
Проверьте нет ли плохого контакта на линии PL (ножки 1 на микросхемах, желтые линии на схеме) и на линии от ножки 10 первого регистра на ножку 9 второго (оранжевый цвет на схеме). 
Возможны глюки, если второй регистр в цепочке отсутствует, но у вас как я понимаю все четыре на месте.

lazenyuk-dmitry, 2022-09-12:
А PL SCK никак не прозвонить мультиметром? Начал прозванивать провода, да вроде все нормально... Не понятно в чем дело.

vsulako, 2022-09-12:
К сожалению, мультиметром  разве что только прозвонить. А посмотреть что там происходит - это нужно осциллограф с памятью, а лучше логический анализатор.
Кнопки 1-16, когда подключены отдельно, работают правильно?

lazenyuk-dmitry, 2022-09-13:
Я вообще не понимаю в чём дело. Все правильно вроде. 

[image]
[image]

Вот так вот все у меня распаяно на регистрах. Больше вроде и путаться нечему. 

Все нормально работает но вот кнопки 25-32 ни в какую. Бесовщина какая то.

lazenyuk-dmitry, 2022-09-13:
> Кнопки 1-16, когда подключены отдельно, работают правильно? 

1-16 все правильно... А не может быть то что у меня длинный кабель до КПП? 130 см кабель. Может конденсатор нужно больше припаять? Или что? Я уже не знаю.

lazenyuk-dmitry, 2022-09-13:
Попробовал конденсатор больше... Такая же фигня.

vsulako, 2022-09-13:
Да, 104 это 100нф. Его должно быть вполне достаточно. 
Впрочем для проверки можно попробовать подключить проводами покороче.
Именно 25-32 не работают? А 17-24 как?
Попробуйте подключить заведомо рабочую цепочку, которые у вас сейчас на 1-16 на место 17-32, и посмотреть как будет работать. Вроде у вас там разъем на них есть так что это должно быть несложно.

lazenyuk-dmitry, 2022-09-13:
> Именно 25-32 не работают? А 17-24 как?

Да вот пока не трогаю 25-32 все нормально вроде. Например зажимаю 18-19 все норм. Жму 27 гаснут 18-19 и загораются 27, 29-30.

lazenyuk-dmitry, 2022-09-13:
> которые у вас сейчас на 1-16 на место 17-32

Это блин сложно будет... Я уже поменял регистры на 17-32 кнопках.

lazenyuk-dmitry, 2022-09-13:
Да нет... Вроде не может быть такого я уже и кнопки отпаял. Причем если кнопки 17-24 не трогать то 25-32 работают нормально...

lazenyuk-dmitry, 2022-09-13:
Поменял местами линии data1 и data2 результаты такие же только наоборот. Теперь глюки на кнопках 1-16

lazenyuk-dmitry, 2022-09-13:
> И впрямь непонятно.
> А посмотрите мультиметром, что на ногах у регистров когда это происходит.
> Зажать 18, 19, 27, чтобы 18-19 погасли и загорелись 27,29,30 и посмотреть какое напряжение на ножках соответствующих 18,19,27,29,30. По идее на тех кнопках что действительно нажаты должно быть 0, а на ненажатых около +5.  Вряд ли будет иначе но вдруг.

Да все так и есть как и ожидается...

vsulako, 2022-09-13:
Выходит, глюки привязаны именно к этой цепочке регистров, что-то там не так.
Насчет длины кабеля - непонятно. По идее не должно так быть, но надо пробовать, подключите проводами покороче.
Может, все разобрать и снова собрать? Поменяв местами регистры например

lazenyuk-dmitry, 2022-09-13:
Нашел я кажется в чем дело.... От части вы были правы... Это провод PL. Я кинул перемычку 20см от базы до КПП на PL линию и все вроде стало работать корректно. 

Что делать?

vsulako, 2022-09-13:
То есть если подключить другими проводами то все работает? Может, с длинным кабелем что-то не так, один провод скажем перебит где-то?

lazenyuk-dmitry, 2022-09-13:
> То есть если подключить другими проводами то все работает? Может, с длинным кабелем что-то не так, один провод скажем перебит где-то? 

Сомневаюсь.. я использую провода от интернет кабеля покупал в магазине. И я их уже звонил. Я только на PL сделал перемычку и все. Может не хватает мощи?

Щас прозвонил провод на PL все ок. Выходит дело в длине.

vsulako, 2022-09-13:
Странно что SCK работает а PL нет.  А сопротивления у всех жил одинаковые? она может звониться но иметь большое сопротивление. Если там есть запасные жилы можно другую взять, или несколько в параллель. 
И еще непонятно почему глючит именно после нажатия кнопки.  Сопротивления резисторов на кнопках какие?

vsulako, 2022-09-13:
Проверьте заодно и эти резисторы, вдруг там какой бракованный попался.

lazenyuk-dmitry, 2022-09-13:
> Проверьте заодно и эти резисторы, вдруг там какой бракованный попался. 

Проверял. На кнопках резисторы 8.9, 9.15, 10. Ну это я думаю погрешность. И норма. Наверно)

Провода жилы что PL что SCK 1ом сопротивление. Не в силах дело.

lazenyuk-dmitry, 2022-09-13:
Нельзя там просто подтяжку сделать? Или что?

vsulako, 2022-09-13:
Да попробовать можно, только не уверен что поможет. Еще один резистор 1-10к с PL регистра к +5в. Хуже быть не должно, во всяком случае.

lazenyuk-dmitry, 2022-09-13:
> Да попробовать можно, только не уверен что поможет. Еще один резистор 1-10к с PL регистра к +5в. Хуже быть не должно, во всяком случае. 
> 
> а так-то можно вообще обойтись без PL, поставив еще резистор, диод и конденсатор. 

Блин ну кажется все работает. На 10к не помог а на 1к все пошло. Можно даже не на сами регистры а на линию PL прицепить подтяжку. А как подобрать сопротивление? Или это собо ни на что не влияет? 

На SCK тоже можно подтяжку сделать? Я погугли немного вроде подтяжки можно и на PL и SCK сделать. Что то похожее находил. Можете это узнать? Если что можно ссылочку или картинку. 

Я на днях буду все в порядок приводить. 

Спасибо.

lazenyuk-dmitry, 2022-09-13:
И не плохо было бы добавить подтяжки в документацию.

vsulako, 2022-09-13:
Можно и на SCK. Точно так же - через резистор к 5в. Хуже не будет. С 1к работает - ну и ладно, чего там еще подбирать. обычно для сигналов годится что угодно в районе 1-50к (выше уже помехи может ловить, а ниже - потребление становится ощутимым)
Вообще я в недоумении. Подтяжки там не должны требоваться нигде кроме кнопок, выход ардуино должен без проблем справляться. Попробовал у себя на двухметровом кабеле - все работает. Правда, у меня регистры в SMD исполнении, но по моему разницы быть не должно. И PL с SCK сигнальные входы, потреблять должны самый мизер. Не понимаю.

lazenyuk-dmitry, 2022-09-13:
Я тоже не знаю почему так) но ещё все соберу с подтяжками и протестирую.

lazenyuk-dmitry, 2022-09-13:
Пишут вроде что на каких то регистрах есть внутри уже подтяжки) я хз я не сильно электронщик.

lazenyuk-dmitry, 2022-09-15:
У меня продолжается какая то фигня с кнопками.... Я уже не знаю что делать с ними. Подпаял подтяжки припаял кнопки. Глючат по прежнему. Половина кнопок нормально работает а половина гонит. Поэтому мне показалось что работают. Но на самом деле нет.

vsulako, 2022-09-15:
может, это микросхема? что если их поменять местами?

lazenyuk-dmitry, 2022-09-15:
Уже нету идей никаких

vsulako, 2022-09-15:
Одна цепочка работает а вторая нет.  надо искать чем вторая отличается от первой.
Подключить короткими проводами чтобы исключить влияние кабеля.
временно убрать второй регистр в цепочке, ногу 10 первого притянуть резистором к VCC. Должны работать первые 8 кнопок.
потом подключить второй регистр, без кнопок. все также должны работать первые 8, а вторые 8 - не работать.
потом по одной добавлять кнопки.

lazenyuk-dmitry, 2022-09-15:
> Одна цепочка работает а вторая нет.  надо искать чем вторая отличается от первой.
> Подключить короткими проводами чтобы исключить влияние кабеля.
> временно убрать второй регистр в цепочке, ногу 10 [...]

А вротой регистр подпаять с подтяжками на кнопки?

vsulako, 2022-09-15:
Наверное лучше сначала без, а потом их постепенно добавлять. Тогда эти кнопки без подтяжек будут видеться как нажатые.

lazenyuk-dmitry, 2022-09-15:
Первый регистр в цепочке работает нормально. Подключаю второй на втором включены все кнопки. Тыкаю подтяжку на 3 ногу и все кнопки гаснут, на двух регистрах. И вообще как то второй регистр живёт своей жизнью.

lazenyuk-dmitry, 2022-09-15:
Вообще все как гирлянда мигает на втором гегистре

vsulako, 2022-09-15:
А есть возможность заменить второй полностью? вместе с резисторами, проводами - чтобы все другое?

lazenyuk-dmitry, 2022-09-15:
> А есть возможность заменить второй полностью? вместе с резисторами, проводами - чтобы все другое?

Да. Но я его менял до этого в паре с первым. Почему то даже просто пальцами до ножек дотрагиваясь начинается какая то не понятная фигня.

vsulako, 2022-09-15:
Без подтяжек реагирование на пальцы - это нормально, для того там подтяжки и нужны.

lazenyuk-dmitry, 2022-09-15:
Все так же и с другим регистром. Только цепляю подтяжку на 3 ногу. Гаснут пару кнопок на втором регистре и те что были включены на первом тоже гаснут. Бесовщина какая-то.

vsulako, 2022-09-15:
А до этого первый регистр и другая цепочка работают как положено? отрабатывают все кнопки в любых комбинациях
и только после того как прицепляется подтяжка ко второму регистру, даже без нажатия соответствующей кнопки начинаются глюки?

lazenyuk-dmitry, 2022-09-15:
> А до этого первый регистр и другая цепочка работают как положено? отрабатывают все кнопки в любых комбинациях
> и только после того как прицепляется подтяжка ко второму регистру, даже без нажатия соответствующей кнопки начинаются глюки?
> 

Да все регистры кроме последнего работают хорошо. Постедний даже если без подтяжек подключить и проживать на остальных кнопки, то на последнем мигают кнопки. Но остальные при этом работают отлично.

vsulako, 2022-09-15:
Очень странно. С подтяжками что-то не так, что ли? 
Ну а если его подключить без подтяжек и на входы для кнопок подавать GND или VCC?
GND должно включать соответствующую кнопку а VCC гасить. Остальные кнопки на нем могут моргать, но та к которой подключается - должна работать, и не влиять на кнопки на других регистрах.

lazenyuk-dmitry, 2022-09-15:
> Очень странно. С подтяжками что-то не так, что ли?
> Ну а если его подключить без подтяжек и на входы для кнопок подавать GND или VCC?
> GND должно включать соответствующую кнопку а VCC гасить. [...]

То же самое. Подаю vcc так же отрубает кнопки другого регистра. Без понятия что делать.

vsulako, 2022-09-15:
Я тоже... 
Эксперимент: отсоединить от второго регистра поочередно PL, SCK  и ту что идет к 9й ноге. Будет ли также отрубать кнопки у первого?

lazenyuk-dmitry, 2022-09-15:
Смотри... Почему то иногда нормально работает второй регистр, а  иногда глючит, как будто зависит от того как его держишь. Как будто какие то помехи. 

И есть такой глюк гадание 29-32 тригерит 25-28, то есть нажатие 29 так же нажимает 25 и так далее. Как будто один ряд на регистре тригерит второй. А иногда все нормально работает. Что такое не понимаю.

lazenyuk-dmitry, 2022-09-15:
То вообще на втором регистре часть кнопок не срабатывает то работает. Не понимаю нифига в чем проблема. Может какие помехи ловит?

lazenyuk-dmitry, 2022-09-15:
Я не совсем разобрался но там вроде бы есть вроде как цикл который опрашивает регистры. Так вот я читал что стоит делать паузу в опросах небольшую 40мс где то чтобы фильтровать помехи то ли что. У вас это реализовано? Я просто не знаю уже в чем дело.

vsulako, 2022-09-16:
Если зависит от того как держать то это похоже на плохой контакт. 
Паузы там вообще ни к чему. Это разве что против дребезга на кнопках, но тут явно не то.  У меня работает без пауз, да и у вас три из четырех работают. Очевидно же что проблемный этот четвертый. Если бы была проблема в коде - она проявлялась бы на обеих цепочках. Ну или не переезжала бы при замене цепочек местами.
Пересоберите всю эту часть заново из всего нового, включая провода. Есть же первая цепочка которая работает правильно, вот точно такую же надо сделать вторую. Конденсатор можно пока убрать.

lazenyuk-dmitry, 2022-09-16:
В чем может быть дело? Я дотрагиваясь до провода PL и все работает тогда.

lazenyuk-dmitry, 2022-09-16:
Блин похоже что дело в кабеле. У меня кабель от интернета и концевики RJ45, кабель 8 жил проволоки медной. Заменил на кабель от usb простой. И все работает. Ну и он по короче. Что получается что кабель не подходит почему то? А у меня так все хорошо складывалось на этих кабелях и разъёмах... Обидно... Теперь что то переделывать нужно. На интернетовской кабеле работают педали и ручник нормально и остальные оси тоже. У вас какой кабель используется? Мне нужно 8 жил на КПП.

lazenyuk-dmitry, 2022-09-16:
Ты можешь объяснить почему на витой паре не работает а с обычным usb кабелем работает? Я не понимаю... Вроде провод да провод чего такого...

lazenyuk-dmitry, 2022-09-16:
Странно... Если по два провода витой пары пустить на PL SCK DATA вместо одного то тоже все работает.

Я даже так предполагаю что хватит два провода только на PL и все будет работать. Может у тебя есть идеи по этому поводу) дичь какая то.

vsulako, 2022-09-17:
У меня кабель телефонный плоский, КСПВ 4х0.4, и телефонные разъемы RJ12. 
Восьмижильная витая пару UTP5 показалась слишком жесткой.
Причин чтобы на UTP5 не работало не вижу вообще никаких. Кроме подозрений что одна из жил повреждена.

lazenyuk-dmitry, 2022-09-17:
> У меня кабель телефонный плоский, КСПВ 4х0.4, и телефонные разъемы RJ12. 
> Восьмижильная витая пару UTP5 показалась слишком жесткой.
> Причин чтобы на UTP5 не работало не вижу вообще никаких. Кроме подозрений что одна из жил повреждена. 
> 

Я уже менял провод. Оно то работает но глючит.

lazenyuk-dmitry, 2022-09-18:
Вообще интересно как то... Подключаю DATA и PL на одну пару, то есть голубой и бело-голубой. И у меня глючат кнопки но стоит только перенести подключение на разные витые пары так все отлично работает. Это какие то помехи?

lazenyuk-dmitry, 2022-09-18:
Я тут уже начал фигнёй страдать. Подрубаю сопротивление 1к 10к в разрыв линии PL и все работает. Ещё можно и на DATA кинуть в разрыв сопротивление. Я не понимаю почему но работает. Думаю даже так и оставить. 

Василий, может быть напишем научную работу по дичи с витой парой? 😄

lazenyuk-dmitry, 2022-09-18:
Короче установил сопротивление 1к в разрыв PL, все собрал и все работает. Странно. Очень интересно но ничего не понятно. 

Подскажи форум электронщиков где можно задать такой вопрос. Интересно есть ли разумное объяснение этому.

vsulako, 2022-09-18:
Мне тоже непонятно. Посмотреть бы что там происходит, но нечем. И сам потыкать не могу.
форумы.. ну наверное radiokot.ru easyelectronics.ru cxem.net

[1 more reply]

closed 2022-09-18 by lazenyuk-dmitry