vsulako/AFFBWheel #26: Подключение двух двигателей на один вал.

lazenyuk-dmitry, 2022-09-14:
@vsulako Василий снова привет) я что то задался все же вопросом правильно ли я подключил два двигателя. У меня просто они смотанны параллельно и подключены к BTS. Вопрос, работают ли они на полную? Как это можно проверить? И может быть есть варианты подключения более грамотные? Например драйвер мотора с выходами на два двигателя?

Наверно можно замерить ампераж или что на моторах под нагрузкой?

lazenyuk-dmitry, 2022-09-14:
Было бы хорошо если можно снять больше мощности с моей конструкции)

vsulako, 2022-09-14:
К сожалению, тут не могу сказать, не знаю. У меня двигатель один и тот слабый. По моему, параллельного подключения достаточно. Не думаю что BTS чем-то может ограничивать. 
Ну можно наверное сравнить как работает с BTS и если напрямую подключить моторы к БП. Измерить ток через каждый мотор.

lazenyuk-dmitry, 2022-09-15:
Кстати, я заказывал вот здесь. 

https://sl.aliexpress.ru/p?key=g8CNF2

3000 оборотов 24В блока питания на 10А вроде хватает, все работает. Вышло не дорого два мотора.

Говорят вроде лучше брать меньше оборотов для руля.

vsulako, 2022-09-15:
Меньше оборотов - видимо, чтобы не городить чересчур большой или многоступенчатый редуктор.
Судя по табличке в описании, 775 на 24в под обычной нагрузкой жрет 2А, а в пике при заблокированном роторе -13А. Как в реальности - неизвестно. 
Получается в пике 26А на два двигателя, БП может не вытащить, да и контакты на BTS поплавиться могут. 
Может, имело бы смысл добавить еще один такой же БП, и заодно еще один BTS, по одному на каждый двигатель. 
Еще тут вопрос - возможно ли будет удержать такой руль, он еще вырвется да чего-нибудь поломает, ладно если свой вал, а если пальцы? 

Вообще это тема для исследований - два мотора на один вал. 
https://electronix.ru/forum/index.php?app=forums&module=forums&controller=topic&id=122546#comments
Вроде как при этом один мотор получается ведущий а другой ведомый, и часть мощности может теряться.
Зачем-то предлагают добавлять подбираемое сопротивление на один из двигателей, или соединять их последовательно. Даст ли это что-нибудь - не знаю.

lazenyuk-dmitry, 2022-09-18:
> Меньше оборотов - видимо, чтобы не городить чересчур большой или многоступенчатый редуктор.
> Судя по табличке в описании, 775 на 24в под обычной нагрузкой жрет 2А, а в пике при заблокированном роторе [...]

Скорее всего советую последовательно подключить потому что для последовательного соединения I=I1=I2 то есть типа сила тока на каждом моторе должна быть одинаковая и максимальная, в теории. Но напряжение делиться U=U1+U2. 

Ну вроде у меня, ни блок, ничего не греется, ни соединения. Думаю ничего пока делать не буду. Ну я смотрел вроде таблицу для двигателя 3000 оборотов 24В там 3.2А потребление. Поэтому блок на 10 взял. Да и большее что я нашел это блок на 10А я взял модель блока типа от ноутбука чтобы все компактно и аккуратно было.

closed 2022-09-20 by lazenyuk-dmitry