vsulako/AFFBWheel #27: База директ драйв с мотор колесом от гироскутера.

Iurii-Dolgorukii, 2022-09-20:
Добрый день. Перейду сразу к делу)))
Собрал базу ДД использовав мотор колесо от гироскутера. Работает, вроде не плохо. Но у меня две платы стм, на одной прошивка ВНМ, другая управляет мотором. Управляющий сигнал аналоговый, имеет много шумов, да и не хорошо это, два мк тратить на это дело. 
Вопрос в следующем:
-Можно ли эту прошивку применить на стм32? (для тестов имеется Ф401, 407, 411. Ф103с8т6).
-если убрать из прошивки все оси, алгоритм управления бтс, оставив только ось руля, на сколько меньше памяти в ардуино она станет занимать? (библиотека, которую я применил для бк мотора, занимает тоже много места).
ну и кратенькое видео)))
баранка блютуз, с есп32
https://youtu.be/760Q2IoW_f8

vsulako, 2022-09-20:
- на STM32 не получится.
- думаю, можно освободить минимум треть флеша(около 10к), а может и больше.

Iurii-Dolgorukii, 2022-09-20:
А можете подсказать, где именно распологаются данные для направления руля? Я не силён в программировании, по этому сложно разобраться, что именно в Вашей прошивке я должен оставить))

vsulako, 2022-09-20:
Что такое - данные для направления руля?

Вот тут: https://github.com/vsulako/AFFBWheel/blob/master/AFFBWheel/AFFBWheel.ino

В строке 224 начинается основной цикл.
Данные о положении руля передаются в строке 246 -  `wheel.axisWheel->setValue(GET_WHEEL_POS);`
(GET_WHEEL_POS определяется ранее в зависимости от используемого датчика положения)
`readAnalogAxes();` и `readButtons();` - это опрос других осей и кнопок, можно убрать.
`processUsbCmd();` - работа с GUI.
Из важного остается `wheel.update();` - это обмен данных по USB с компьютером, и `processFFB();` - обработка обратной связи.
Далее  `processSerial();` - обработка команд из компорта.

Управление двигателем находится в https://github.com/vsulako/AFFBWheel/blob/master/AFFBWheel/motor.cpp

vsulako, 2022-09-20:
Что у вас за библиотека для мотора?

Iurii-Dolgorukii, 2022-09-20:
Мне сложно сформулировать. 
для управления мотором у меня переменная (int S;), максимальное усилие равно -50/50. вращение влево/вправо.
[code]
    motor.move(S);

[/code]
можно из вашего кода получить информацию подобного рода?))

Iurii-Dolgorukii, 2022-09-20:
но может возникнуть проблема с энкодером наверно, этой библиотеке как и вашей он нужен. Возможен кофликт?

lazenyuk-dmitry, 2022-09-20:
@vsulako, у него библиотека для управления мотором эта https://simplefoc.com/ она поддерживает STM и говорит там она всего 20 процентов а на ардуино 70 вроде занимает. 

Я нашел такую библиотеку, кто то перенес библиотеку для ардуино на СТМ, https://github.com/billydragon/STM32_FFBJoystick думаю можно по быстрому склепать из двух библиотек прошивку) 

Вожножно можно договориться с Василием если ему это интересно)

Iurii-Dolgorukii, 2022-09-20:
спасибо, буду пробовать)

vsulako, 2022-09-20:
> Возможен кофликт?
С бесколлекторными двигателями не работал, но как я понимаю, там нужно знать абсолютное положение оси руля? 
Возможно, придется чтото переписывать. 
Может, будет проще взять за основу нечто вроде вот этого? https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

lazenyuk-dmitry, 2022-09-20:
> > Возможен кофликт?
> С бесколлекторными двигателями не работал, но как я понимаю, там нужно знать абсолютное положение оси руля? 
> Возможно, придется чтото переписывать. 
> Может, будет проще взять за основу нечто вроде вот этого? https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

Эта библиотека как раз есть под СТМ я кинул ссылку выше.

Iurii-Dolgorukii, 2022-09-20:
при первом включении контроллер вращает двигатель параллельно считывая энкодер, для определения электрического угла. Далее работает по командам.

Iurii-Dolgorukii, 2022-09-20:
> > Возможен кофликт?
> > С бесколлекторными двигателями не работал, но как я понимаю, там нужно знать абсолютное положение оси руля?
> > Возможно, придется чтото переписывать.
> > Может, будет проще взять за основу нечто вроде вот этого? https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

она выглядит вроде просто, но куда энкодер подключать?)) игре он ведь нужен

Iurii-Dolgorukii, 2022-09-20:
> Что у вас за библиотека для мотора?

https://simplefoc.com/

lazenyuk-dmitry, 2022-09-20:
> она выглядит вроде просто, но куда энкодер подключать?)) игре он ведь нужен

Наверно через библиотеку. https://github.com/bubulindo/STM32Encoder типа этой.

lazenyuk-dmitry, 2022-09-20:
И кстати если у вас получиться можно выложить проект на гитхаб. Будет полезно.

Iurii-Dolgorukii, 2022-09-20:
Попробую как и раньше, методом тыка))))
Если будут идеи, то мой стенд всегда готов к испытаниям. Микроконтроллеры тоже

vsulako, 2022-09-20:
> она выглядит вроде просто, но куда энкодер подключать?)) игре он ведь нужен
Это же только основа. Энкодер или другой датчик надо добавлять вручную, там есть примеры.

Хотя наверное все равно ничего не выйдет. Попробовал собрать минимальный скетч с библиотекой SimpleFOC - заняло 89% флеша. Не влезет.

lazenyuk-dmitry, 2022-09-20:
> Это же только основа. Энкодер или другой датчик надо добавлять вручную, там есть примеры.
>
> Хотя наверное все равно ничего не выйдет. Попробовал собрать минимальный скетч с библиотекой SimpleFOC - [...]

Это на ардуино? А у тебя нету СТМ?

Iurii-Dolgorukii, 2022-09-20:
На стм бы влезло я думаю...

vsulako, 2022-09-20:
Вот с STM к сожалению не знаком. Валяются пара F103С8T6, но пока до них не добрался чтобы поизучать :(

Iurii-Dolgorukii, 2022-09-20:
> И кстати если у вас получиться можно выложить проект на гитхаб. Будет полезно.

Это разумеется

Iurii-Dolgorukii, 2022-09-20:
> Это же только основа. Энкодер или другой датчик надо добавлять вручную, там есть примеры.
>
> Хотя наверное все равно ничего не выйдет. Попробовал собрать минимальный скетч с библиотекой SimpleFOC - [...]

А что значит минимальный?
Simplefoc считывает энкодер, у ребят надо спросить, может можно вывести данные энкодера, чтобы не подключать ещё одну библиотеку

vsulako, 2022-09-20:
> А что значит минимальный?
Только то что описано тут: https://docs.simplefoc.com/code
и то пришлось убрать датчик тока.

lazenyuk-dmitry, 2022-09-20:
Я попробую что нибудь написать под СТМ на библиотеках) правда никогда не писал на плюсах. Ну может быть что то получиться с библиотеками. Стало чёт интересно поковыряться)

Просто если что можно объединить усилия если кто то тоже думает склепать проектик)

Iurii-Dolgorukii, 2022-09-20:
> Это же только основа. Энкодер или другой датчик надо добавлять вручную, там есть примеры.
>
> Хотя наверное все равно ничего не выйдет. Попробовал собрать минимальный скетч с библиотекой SimpleFOC - [...]

Там если за комментировать все строчки связанные с serial Port, то можно немного памяти сэкономить

Iurii-Dolgorukii, 2022-09-20:
У меня занимает 74 процента.

vsulako, 2022-09-20:
> Там если за комментировать все строчки связанные с serial Port, то можно немного памяти сэкономить

```
#include <SimpleFOC.h>

BLDCMotor motor = BLDCMotor(11, 9.75);
// Encoder(pin_A, pin_B, PPR)
Encoder sensor = Encoder(2, 3, 2048);
// channel A and B callbacks
void doA(){sensor.handleA();}
void doB(){sensor.handleB();} 
// instantiate motor
BLDCDriver3PWM driver = BLDCDriver3PWM(9, 5, 6, 8);

// instantiate sensor 
// instantiate current sensor   
//InlineCurrentSense current_sense = InlineCurrentSense(0.01, 50, A0, A2);

void setup() {  

    sensor.init();
  // hardware interrupt enable
  sensor.enableInterrupts(doA, doB);

  //current_sense.linkDriver(&driver);
  // init current sense
  //current_sense.init();

  // init sensor
  // link motor and sensor
  
// pwm frequency to be used [Hz]
  driver.pwm_frequency = 20000;
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  // Max DC voltage allowed - default voltage_power_supply
  driver.voltage_limit = 12;
  // driver init
  driver.init();

  // link motor and driver
  // link driver and the current sense

  // link motor and current sense
  motor.linkSensor(&sensor);

  // init driver
  // link the motor to the driver
  motor.linkDriver(&driver);
  // link driver and the current sense
  
  // link the motor to current sense
//  motor.linkCurrentSense(&current_sense);

  // set control loop type to be used
  motor.controller = MotionControlType::velocity;
  // initialize motor
  motor.init();
  

  // init current sense

  // align encoder and start FOC
  motor.initFOC();
}

void loop() {
  // FOC algorithm function
  motor.loopFOC();

  // velocity control loop function
  // setting the target velocity or 2rad/s
  motor.move(2);
}
```
Вот это я собирал. Энкодер, мотор, драйвер, больше ничего. И все равно 89%.
Оно конечно можно вынести все это в отдельную ардуину и с ней как-то общаться, но у вас и сейчас как я понимаю то же самое.

Iurii-Dolgorukii, 2022-09-21:
Да, тоже самое. Но через analogRead. А он шумит зараза. В нулевой точке нет отключения драйвера стабильного. Мотор дрожит. А если канал отключения драйвера исключить, то тяжело вращать мотор. Так как мосфеты нижних плечей открыты и обмотки мотора между собой замкнуты

Iurii-Dolgorukii, 2022-09-21:
А вы можете из прошивки вашей исключить всё лишнее, сколько она интересно займёт

Iurii-Dolgorukii, 2022-09-21:
Кстати я датчик тока не использую, я подобрал значение в Limit. voltage при котором максимальное потребление не привыкает 4 ампера. У меня бп при 4,5 в защиту уходит))))

Iurii-Dolgorukii, 2022-09-21:
// pwm frequency to be used [Hz] driver.pwm_frequency = 20000;
На ардуино частота фиксированная, эта строчка не сработает

vsulako, 2022-09-21:
> Да, тоже самое. Но через analogRead. А он шумит зараза. В нулевой точке нет отключения драйвера стабильного. Мотор дрожит. А если канал отключения драйвера исключить, то тяжело вращать мотор. Так как мосфеты нижних плечей открыты и обмотки мотора между собой замкнуты

Хм. А зачем через analogRead? Прошивка с контроллером закрыта и приходится тупо измерять что она выдает на канал управления двигателем? Что такое прошивка ВНМ мне чего-то найти не удалось.
По идее если на выходе сигнал ШИМ, то можно его анализировать и получать более точное значение.

> А вы можете из прошивки вашей исключить всё лишнее, сколько она интересно займёт

могу, но это надо повозиться. Мне кажется даже если оставить только самое основное, будет не меньше 35-40%.
Но вариант с двумя контроллерами мне кажется можно было бы сделать. Со связью не аналоговым сигналом а скажем через SPI. Туда значение усилия для мотора, обратно данные с энкодера. Второй контроллер может быть каким-нибудь Uno или Nano - для них кстати SimpleFOC меньше получается, около 70%.

vsulako, 2022-09-21:
Ориентировочно:
Если оставить только ось руля и ОС - получается около 45%.
Если еще порезать ОС до одного эффекта - около 35% 
В лучшем случае, думаю, еще до 30% можно довести.

Iurii-Dolgorukii, 2022-09-21:
Да, прошивка закрыта.
https://github.com/hoantv/CarSimulatorFirmware
Я не знаю, получится ли получить точные значения не теряя в скорости. По выходному pwm работает у меня

vsulako, 2022-09-21:
Зависит от частоты. Будет задержка в один период, т.к. определить значение можно только после окончания периода. И чем выше частота, тем больше будет потеря в точности.

lazenyuk-dmitry, 2022-09-21:
> могу, но это надо повозиться. Мне кажется даже если оставить только самое основное, будет не меньше 35-40%.
> Но вариант с двумя контроллерами мне кажется можно было бы сделать. Со связью не [...]

Может быть это разумнее решение будет? Если что то новое пилить это много очень работы будет. Плюс еще бы хорошо сделать GUI для всего этого дела. А тут все готовое. Тем более если Василий готов помочь. Это только написать проект для управления двигателем получается нужно? Ну и допилить прошивку AFFBWheel чтобы она могла взаимодействовать со вторым МК? Так получается? И это наверно будет лучше. Уже есть готовый проверенный проект, оттестирован. 

@Iurii-Dolgorukii И что у вас с педалями? Они не они работают как отдельное устройство? Кнопки на руле я так понимаю работают отдельно от базы?

Чтобы для STM что то годное делать научиться это наверно только учиться пол года нужно) Так что вот такие мои мысли. Это будет надежнее, две платы ардуино и нормальное общение между ними реализовать, и получите все фишки и плюшки данной прошивки и так же какие то обновления и какаю никакую поддержку.  

Если урезать прошивку все равно не поможет то и не нужно)

Iurii-Dolgorukii, 2022-09-21:
> > могу, но это надо повозиться. Мне кажется даже если оставить только самое основное, будет не меньше 35-40%.
> > Но вариант с двумя контроллерами мне кажется можно было бы сделать. Со связью не аналоговым сигналом а скажем через SPI. Туда значение усилия для мотора, обратно данные с энкодера. Второй контроллер может быть каким-нибудь Uno или Nano - для них кстати SimpleFOC меньше получается, около 70%.
> 
> Может быть это разумнее решение будет? Если что то новое пилить это много очень работы будет. Плюс еще бы хорошо сделать GUI для всего этого дела. А тут все готовое. Тем более если Василий готов помочь. Это только написать проект для управления двигателем получается нужно? Ну и допилить прошивку AFFBWheel чтобы она могла взаимодействовать со вторым МК? Так получается? И это наверно будет лучше. Уже есть готовый проверенный проект, оттестирован. 
> 
> @Iurii-Dolgorukii И что у вас с педалями? Они не они работают как отдельное устройство? Кнопки на руле я так понимаю работают отдельно от базы?
> 
> Чтобы для STM что то годное делать научиться это наверно только учиться пол года нужно) Так что вот такие мои мысли. Это будет надежнее, две платы ардуино и нормальное общение между ними реализовать, и получите все фишки и плюшки данной прошивки и так же какие то обновления и какаю никакую поддержку.  
> 
> Если урезать прошивку все равно не поможет то и не нужно)
> 
> Проекты под stm32 как бы есть для реализации руля но они полумертвые какие то. 

Ну оно у меня так и работает на двух мк. Педали, кнопки на руле это не проблема, можно отдельно на про микро реализовать. В управлении двигателем ничего допиливать не надо, библиотека работает из коробки)))
Ей точные данные необходимо подавать

Iurii-Dolgorukii, 2022-09-21:
Попробую через spi, у гайвера на сайте вчера попадалась библиотека с примерами

lazenyuk-dmitry, 2022-09-21:
> Ну оно у меня так и работает на двух мк. Педали, кнопки на руле это не проблема, можно отдельно на про микро реализовать. В управлении двигателем ничего допиливать не надо, библиотека работает из коробки)))\nЕй точные данные необходимо подавать

Ну просто можно все сделать лучше. Добиться лучшей работы.

Iurii-Dolgorukii, 2022-09-21:
Не просто можно, а необходимо)))

lazenyuk-dmitry, 2022-09-21:
Вся надежда только на Василия если хочешь все сделать хорошо и быстро.

lazenyuk-dmitry, 2022-09-23:
https://github.com/o-devices/bldc-ffb-wheel 

Нашел такой проект.

Iurii-Dolgorukii, 2022-09-24:
> [https://github.com/o-devices/bldc-ffb-колесо](https://github.com/o-devices/bldc-ffb-wheel)
> 
> Нашел такой проект.

видел, он тоже закрыт