vsulako/AFFBWheel #34: Пропуск шагов энкодера

neSecret, 2023-04-12:
Использовал в схеме энкодер 600 импульсов на оборот. Соотношение поворотов руля к энкодеру примерно 3.75. При резких поворотах ардуино начинает пропускать шаги энкодера и центр руля сбивается. Если проблема в высоком разрешении энкодера, то как её можно решить без его замены?

vsulako, 2023-04-12:
К сожалению, у меня такого энкодера нет, проверить не могу. 
Но и при таких параметрах должно бы работать нормально. 
Энкодер подключен на пины 0-1? 
Видел ваше сообщение на 3dtoday, вы написали что собрали по схеме автора поста - но у него несколько другая прошивка.
Без замены энкодера... а нельзя ли уменьшить соотношение передачи? Или перенести энкодер на ось руля, чтобы соотношение было 1:1?

neSecret, 2023-04-12:
> К сожалению, у меня такого энкодера нет, проверить не могу. Но и при таких параметрах должно бы работать нормально. Энкодер подключен на пины 0-1? Видел ваше сообщение на 3dtoday, вы написали что [...]

Энкодер подключен на пины 0-1. Я использовал вашу модифицированную прошивку. Нашёл её в группе Logitech MOMO Racing Modifications (ссылка на архив: https://disk.yandex.ru/d/ATp07dvzKeadaQ). Схему подключения и установки энкодера там же.
Может проблема в скорости обработки сигнала в ардуино? Она просто не успевает считывать столько шагов.
Установить на вал руля, к сожалению, не возможно, потому что из него выходят провода к кнопкам :(

neSecret, 2023-04-12:
Хотел взять такой же энкодер, но с меньшим разрешением, и увидел, что цены на них сильно увеличились за последние пару лет...

neSecret, 2023-04-12:
> В той прошивке я не делал работу с энкодером, ее добавил уже другой автор.
> 
> Вот, попробовал навскидку переделать ее под библиотеку Encoder, которую использую я. Проверить в действии не могу, так что попробуйте вы. Для компиляции понадобится установить в Arduino IDE эту библиотеку: https://github.com/PaulStoffregen/Encoder (она есть и в менеджере библиотек, надо просто ее найти и установить) [diy_g25_enc.zip](https://github.com/vsulako/AFFBWheel/files/11212173/diy_g25_enc.zip)

Я правильно понял, что я могу просто залить эту прошивку в плату? Все калибровочные данные в eeprom сохранятся?

vsulako, 2023-04-12:
Ну, должны бы.