vsulako/AFFBWheel #35: servo motors...

cartaio, 2023-05-10:
Hi i work a lot with servo motors and i have one that could be used as a wheelbase... but they are controlled with a DIR+/- pin and a PULSE+/- pin... can your code be adapted? or do i need to write all by myself?

vsulako, 2023-05-10:
I think it can be adapted.
here is branch with quick changes:  https://github.com/vsulako/AFFBWheel/tree/pwmdir 
in config.h uncomment `#define MODE_PWMDIR`
if necessary, dir and pwm logic can be inverted.

cartaio, 2023-05-10:
thanks... i should have an arduino laying around somewhere... i'll try asap and try to make some prototype to share this summer! if it works should be a cheap build... the servo is around 250€ with included drive...

cartaio, 2023-05-29:
Hi just had a couple of hours to dedicate to the project... in pulse and direction i have problem with holding torque... just too high to use it... i think the solution should be to use analog input for torque and a direction pin... same question as above... is it doable or i need to program it my self?

cartaio, 2023-05-29:
so i tried with a manual pwm signal on the analog in and the drive doesn't like it... tried with a potenziometer to regulate from 0 and +5 and it works so i think with a dac should work pretty easly... i should have a mcp4725 somewhere... the direction pin works fortunatly so should be only to send instead of the pwm signal, i need a value to send to the dac to scale from 0 to +5/10 the drive have a parameter to scale it up... so even 5v is ok

vsulako, 2023-05-29:
Sorry, I don't understand. Did not work with servo motors. 
What is "analog input for torque"? Servo allows to control torque with analog signal on separate pin?
If drive does not like PWM, but works with analog, may be a simple capacitor or RC lowpass filter will do?

cartaio, 2023-05-30:
Yes the drive i have has pwm for position control and analog for speed or torque control... i tried using something but as a read online, to have a clean signal and a good force feedback you need a dac unfortunately... if you want i can send you the manual of the drive...

vsulako, 2023-05-30:
I think it is possible to use I2c DAC like MCP4725, but I don't have any to test if it will work with my bitbang I2c implementation. 

I assume, torque value should be equal to pwm value? 
FFB produces only one value - force, there is no separate speed and torque values. So, it is not possible to order "rotate slow, but with high torque", or vise versa - "fast, but with low torque".

vsulako, 2023-05-30:
I updated pwmdir branch to use MCP4725 (according to datasheet, not tested).
Force value is sent to DAC, scaled to 12 bit. 
Try if it works.

cartaio, 2023-05-30:
I made it work with pwm... i found a parameter on the drive that smooth the signal but it works more or less... i think woth the dac should be perfect... i have a strange behavior when everything is off(wheel check and game closed) i get the wheel to bounce left and right, with wheel check set to simple spring i get vibrations around 0 point... is there a parameter for a deadzone?

vsulako, 2023-05-30:
No, wheel axis does not have deadzone. May be force is too strong, and even very low FFB force near center cause axis to move? try to lower effect gain.

cartaio, 2023-05-30:
I will try tomorrow, i 3d printed an adapter for a wheel now and tomorrow with the dac should be better... thank you for your time!

cartaio, 2023-05-31:
the DAC is outputing always 0v despite the force value... i tried a new software with the adafruit library and it works... then i used bb i2c and it does not... i'm trying to find why... i'll keep you updated.

cartaio, 2023-06-01:
no luck for today... i made some research, and i'll try to do something later... some info i found on the dac attached...
[image]
[image]
[image]

vsulako, 2023-06-01:
1. May be i2c address is wrong. I set it as 0x60, but datasheet says, A1/A2 can be factory set to be not zero. Also, not sure about A0, if it is 0 or 1. Did you use same address 0x60 with another library? 
2. another possibility is that MCP4725 cannot work at too high speed.  try to set I2C_DELAY in config.h from 1 to higher value, 10-20 or so, it will slow down bitbang i2c. 
3. last suggestion is to use hardware i2c. unfortunately, there's not enough space left in flash for adding wire.h library, so something must be sacrificed. and it will require to free pin 3.
code for hw can be like this:

```
void writeDAC(uint16_t value)
{
  value &= 0x0FFF; //clear first 4 bits for fast & normal modes
  Wire.begin();
  Wire.setClock(400000);
  Wire.beginTransmission(MCP4725_ADDR);
  Wire.write(value>>8);            
  Wire.write((uint8_t)value);            
  Wire.endTransmission();           
  Wire.end();
}
```

cartaio, 2023-06-02:
Update! I removed part of the serial output in the main program, fitted the code and it works perfectly... now i will adapt my old pedals bolt the wheel and try it in game... the dac remove the noise that the motor had with pwm... do you have some standard parameter to put in the gui for FF? Or 1024 everywhere is a good starting point?... thanks for the help!

vsulako, 2023-06-02:
1024 for FFB effect gains is default starting point. It means coefficient 1:1, i.e.: if force is 50%, output PWM value will be also 50%. 
Most games use only constant force effect.