vsulako/AFFBWheel #36: AS5600 isnt registering position

Highlight8, 2023-06-14:
can you help me the as5600 as you said i did every thing the same but im unable to resolve the issue and i think i connected it correctly too can you show me how you connected it please

im sharing the code which customized with your help
and i had a doubt about where to change the i2c code in config.h but i did what i think was appopriate 

all other things are working fine except this idk if its an issue with the board because it randomly registers some times

[New WinRAR ZIP archive.zip](https://github.com/vsulako/AFFBWheel/files/11745266/New.WinRAR.ZIP.archive.zip)

vsulako, 2023-06-14:
Connection diagram is in readme: https://github.com/vsulako/AFFBWheel/raw/master/images/AS5600.png
It seems that your config is correct, if I2c SDA/SCL pins are 0-1.

Try to increase I2C_DELAY value in config.h, to 10 or 20.
If you have free pins, also try to change I2c pins from 0-1 to another ones.

Highlight8, 2023-06-19:
thanks for the advice i tried it but the as5600 dosnt work with the test code so im assuming its faulty sorry for the issue thankyou very much for replying im very grateful

closed 2023-06-19 by Highlight8