vsulako/AFFBWheel #37: Rotary encoder isnt registering position

Highlight8, 2023-06-19:
So assuming the as5600 was faulty and i bought this rotary encoder 600 ppr but the thing also does not work so im assuming its a fault in the sketch can you please rectify it (i have made correct connections according to the sketch and it still does not work)
[rotary issue.zip](https://github.com/vsulako/AFFBWheel/files/11789979/rotary.issue.zip)

sorry for the trouble im a student and was doing this as a summer project and bought everything out of my strict budget

Highlight8, 2023-06-19:
i also tried the trick with connecting vcc with a Resistor to A and B

[image]
like this

vsulako, 2023-06-20:
Your configuration is correct - I checked it in hardware with encoder connected to 0-1 pins.
If connections are correct and encoder is ok  - I can only suppose that something is wrong with arduino pins. 

It can be checked with a couple of wires by manually simulating encoder signal: without encoder connected, connect pin 0 to GND, connect pin 1 to GND, disconnect pin 0, disconnect pin 1. This sequence should register as encoder step.
If it does not react, may be pins 0-1 are faulty.

Can you move encoder wires phase A/B to pins 2-3 for testing? (it will require disconnecting button pins PL and DATA1 and disabling 74HC165 buttons)
Here is full code for this setup. (buttons are disabled, encoder moved to 2-3) I checked it also in hardware.
[enc_pin2-3.zip](https://github.com/vsulako/AFFBWheel/files/11807510/enc_pin2-3.zip)

Highlight8, 2023-06-28:
yes the encoder does give output when simulated manually

i tried it still doesnt work (the vcc resistor connection)

im confused why everything else works but this doesnt

vsulako, 2023-06-28:
> yes the encoder does give output when simulated manually
ok, so arduino board and code seem to be ok.

> i tried it still doesnt work (the vcc resistor connection)
Try pulling to GND, not to VCC, as in picture above. If possible, use 1kOhm resistors.

> i already tried it with 2 encoders i bought
what is exact model name of these encoders?

Highlight8, 2023-06-29:
>Try pulling to GND, not to VCC, as in picture above. If possible, use 1kOhm resistors.
i did it again with 1k ohm tried it also with vcc

>what is exact model name of these encoders?
YTO6-OP-600B-2M-5-24V

and im sending this link 
(its the same product but i did'nt buy it from here)

https://dientuvietduc.com/product/encoder-yt06-600b-op-2m-600-xung-hai-k%C3%AAnh-ab-5-24v/

vsulako, 2023-06-29:
Looks like ordinary encoder. It should work. I dont understand.

Let's look if it outputs anything at all.
Here is simplest testing sketch.

```
void setup() {
  
  pinMode(0,INPUT_PULLUP);
  pinMode(1,INPUT_PULLUP);
  
  attachInterrupt(digitalPinToInterrupt(0), change, CHANGE);
  attachInterrupt(digitalPinToInterrupt(1), change, CHANGE);
  
  Serial.begin(9600);
}

void change()
{
  Serial.print(digitalRead(0));
  Serial.print(digitalRead(1));
  Serial.println();
}
void loop()
{

}
```
Connect encoder's A/B to 0/1 pins without resistors, power it, upload code, open Serial monitor and slowly rotate encoder.
Normally it should react like that:
[image]

If it does not print anything, add resistors to VCC, restart Serial monitor and try once more.

If it still does not react: 
- remove these 2 lines from sketch "pinMode(0, INPUT_PULLUP);
- upload
- move resistors to GND
- rotate again.

If it still does not react... 

I did a little research in google, and found that some encoders, despite stated minimum voltage 5v, cannot really work with 5v.
They have 7805 regulator inside, which requires at least 6-7v input to produce stable 5v, otherwise it just does not work properly.
Can you power encoder from another source with voltage higher than 5v? of course, in that case encoder's VCC and arduino's VCC must be separate (but GNDs connected).
Also, just to be sure that arduino's pins will not be damaged, connect A/B through resistors ~1kohm

[image]

Highlight8, 2023-06-30:
i also had researched what you found and had previously tried it with 9v but it didnt work 

but then you said it can b done till 12v i got soo excited as your research was correct about the encoders working on a higher voltage i connected a 12v adapter i had lying around with the diagram you had given 

And it started working with the test sketch.. i mean it was giving output

but once i uploaded the afbbwheel sketch it stopped working and the gui wheel was rotating by itself towards the right side

(without my input)

basically
it doesnt respond with the afbbwheel sketch but works with test sketch

Highlight8, 2023-06-30:
Okay so it does respond and works i just connected the A and B to the 12v of the adapter with a 10k resistor 

but idk when turning to left it takes more rotations whil turining to right it takes less rotations
can this be fixed??