vsulako/AFFBWheel #41: Encoder Axis Troubleshooting

tote46, 2023-09-23:
Hello, i having a problem with games reading X axis (encoder axis i guess), both in AFFB GUI and some other apps like WheelCheck, DlView,DxTweak and even  in windows calibration X axis is read, but it seems like games (Automobilista 2 and F1 2023) cant read it for some reason.
Leonardo board and analog inputs are readed and work just fine, i made some pedals with hall sensors and potentiometers and they work fine in both games.
What can i do?

vsulako, 2023-09-24:
If axis works in windows gaming devices properties, game should see it too. (Though, some games like Forza Horizon work only with controllers known by them)
Don't know about F1 2023, but I tried Automobilista 2 - it works.
I chose custom wheel in control scheme tab, and then manually assigned steer left and steer right controls in edit assignments tab.
What can be a bit confusing - just moving wheel a little back and forth was not enough. I had to move axis all the way from center to limit and then back to center (actually, game says it in a sidebar) and only then axis was recognized as "joy axis X".

tote46, 2023-09-24:
Hi, yes i got f1 2023 to work and everything works even ffb works great, automobilista 2 i couldn't but i didnt try again after getting f1 working.
Maybe if i reduce steering angle to idk 180° and then try to assign the X axis its gonna be easier, idk gonna try later