vsulako/AFFBWheel #42: Using a multiplexer for button inputs

Zredy, 2023-11-25:
Hello, I want to use a multiplexer (4 pins for selecting the readable channel + 1 pin that reads the channel) to read the button states from my wheel.
I tried poking around and implemeting something but I can't figure out how the button values are stored in the "d" or "buttons" variable.
Are they stored in a bit fashion or some other way?
Any help would be appreciated :)

vsulako, 2023-11-26:
Every bit of 32-bit variable stores state of button. 1=pressed, 0=not pressed.
0b00000000000000000000000000000001 = button #1 is pressed
0b00000000000000000000000000000010 = button #2 is pressed
and so on.

Zredy, 2023-11-26:
Amazing stuff. I managed to implement it in my own way but this way with bitRead and bitWrite is so much sleeker. I have everything working, from HX711 brake(which im probably going to replace), to reading the angle and writing the force effect to an esp32 that's running FOC code for the stepper! Only thing, I tried Forza Horizon 5, and it reads all the buttons etc but the force feedback doesn't work, so i run Forza Emuwheel to get the feedback working, Have you seen anything similar?

vsulako, 2023-11-26:
In my case, Forza Emuwheel did not work with Forza. The only solution I could find was GIMX.