vsulako/AFFBWheel #44: Analog button

Ertugrulbabaci, 2023-12-25:
Hello. How many buttons can I connect to a single pin in the analog button connection? I want to use 4x4 matrix button connection (8 for gear and 8 for steering wheel). Also, are resistors necessary?

lazenyuk-dmitry, 2023-12-27:
As I understand it, you are talking about such a scheme. 

[image]

I don’t understand at all what the circuit will look like if it is divided into two, 8 buttons each. This will be very difficult. And you will need to run 8 wires to your gearbox.

Now there is a diagram in the description where you can do all this using 3 wires. I don't think it makes sense to add buttons on a matrix circuit to the firmware.

Ertugrulbabaci, 2023-12-27:
The reason I want this system is because I have ready-made connections. And I asked, thinking that maybe there was a suitable scheme in the system you prepared. I don't think about money. Which system can I use the most buttons using the least amount of cables? Which one should I choose?

lazenyuk-dmitry, 2023-12-27:
> The reason I want this system is because I have ready-made connections. And I asked, thinking that maybe there was a suitable scheme in the system you prepared. I don't think about money. Which system can I use the most buttons using the least amount of cables? Which one should I choose?

Ok. Well, I don’t think it will be difficult to redo the wires on your assembly. It would be better and more compact. But you can ask @vsulako to do this if he wants. But again, other connection options are much better.

vsulako, 2023-12-27:
1) There is no limit for analog pin buttons, you can connect up to 32 buttons to single pin. But, this type of connection can not detect two or more pressed buttons, only one at a time, so it is not a good choice for usual buttons, it can be ok for a shifter, because it selects only one gear per time. 
2) Sorry, no button matrix support for now. It required too many pins - 4x4 matrix will take 8 pins. Also, matrix wiring is somewhat messy. I didn't think someone would need it. Will try to add when have some time.
3) I think simplest connection for buttons is with i2c extenders: PCF8575/8574. They are cheap, 4 wires only, no need for extra resistors for every button, can be arranged in 4x8, 2x8+1x16 or 2x16 layouts.

Ertugrulbabaci, 2023-12-27:
thanks for the answer. I will try to use 74HC165 shift register for gear (8 buttons). (I hope I can succeed) I will try to make the 8 buttons on the steering wheel with analog button connection. This seems to be the easiest.

lazenyuk-dmitry, 2023-12-27:
> thanks for the answer. I will try to use 74HC165 shift register for gear (8 buttons). (I hope I can succeed) I will try to make the 8 buttons on the steering wheel with analog button connection. This seems to be the easiest.

You can do everything on analog buttons or everything on registers. Just keep in mind that you won't be able to press multiple buttons on an analog connection. Analog buttons are more suitable for gearboxes.

vsulako, 2023-12-28:
Added support for button matrix.
configuration:
```
#define BM                                //uncomment to enable
#define BM_COL_PINS   2,3,4    //pins for columns, comma-separated
#define BM_ROW_PINS   5,6,7  //pins for rows
#define BM_1ST_BTN    1          //number of first button (1-32)
```

Ertugrulbabaci, 2024-01-04:
Thank you very much. I think the other methods you mentioned make more sense. If I fail to do so, I will go back to the matrix link.

Ertugrulbabaci, 2024-01-05:
Hellı again. I'm trying to make a connection using the analog button method. I made the button connections with 4.7k resistors. I cannot assign buttons because I cannot see the values. How can I see the values? Also, can I use both analog connection and shift register at the same time?

vsulako, 2024-01-05:
- Yes, shift register and analog pin buttons can be used at same time. Analog pin buttons will rewrite corresponding shiftreg buttons.
- Connect to ardiuno with Serial Monitor and send command `apbout`. It will turn on outputting value from analog pin selected in config. (`#define APB` must be uncommented and `APB_PIN` defined)

Ertugrulbabaci, 2024-01-06:
Okay, I took care of the analog buttons. I decided to use matrix connection instead of shift register for gear, but I could not understand the pin connections in the matrix codes. 234 567 says 6 pins in total, but 8 cables are needed for 16 buttons. Should I make the connection according to the diagram in the picture above?

vsulako, 2024-01-06:
You say you already have button matrix.  
With 16 buttons it should have 8 pins, 4 for rows and 4 for columns. Internal connections should be like on a picture above. 
You need 8 free pins on arduino. I dont know which you have, but say for example, they are 2,3,4,5,6,7,A4,A5.   
Just list them in config separated by commas, columns and rows pins separately.

````
#define BM            //enable
#define BM_COL_PINS   2,3,4,5
#define BM_ROW_PINS   6,7,A4,A5
#define BM_1ST_BTN    1
````
And connect to your button matrix accordingly.

Usually rows/cols pins are interchangeable, but sometimes matrix can have diodes inside, to prevent ghosting.
If if does not work, try to swap columns/rows pins (diode's cathode should be at row's pin)

Ertugrulbabaci, 2024-01-06:
I've been trying to find where the problem is for hours. I was connecting the buttons but it wasn't working. You have uploaded a new ino file. I was trying to write the Matrix code myself. When I downloaded the new ino file, my problem was solved. thanks a lot.

Ertugrulbabaci, 2024-01-19:
Hi bros again. After giving direction with hatswitches, can we add a button that we can use instead of backspace and enter? Is the button code available? @vsulako

vsulako, 2024-01-21:
What do you mean by backspace/ESC/enter? It's not a keyboard, it's a gaming controller, it does not have keyboard keys at all.
If you want to emulate keyboard button press, you can use software like [JoyToKey](https://joytokey.net/en/) and [similar](https://alternativeto.net/software/joytokey/).

Ertugrulbabaci, 2024-01-22:
For example, you entered the settings tab in the menu. I meant a button you can use when you want to go back to the main menu tab. Like the X and O buttons on the playstation controller. Is this possible?

vsulako, 2024-01-23:
Game should allow to define which controller button does actions like navigating in menus.  
If not, just use a mapper to map buttons to esc/backspace/enter/whatever.  
[image]

SoullessHax, 2025-05-29:
Hello everyone, I wanted to ask a question, there is no easier way to connect the buttons, I saw in another project that analog connectors and ordinary ones are simply used without power and grounding. I just want to temporarily change gears and that's it.

SoullessHax, 2025-05-29:
Нельзя сделать что то подобное
void G25_::setButtons(uint32_t btns)
{
   
  //первые 4 бита - кнопки крестовины (hat switch)
  uint8_t hat=(btns & 0xF);
  btns &= 0xFFFFFFF0;
  switch (hat)
  {
     case 0b00000001:
        btns |= HAT_UP;
        break;
     case 0b00000010:
        btns |= HAT_DOWN;
        break;
     case 0b00000100:
        btns |= HAT_LEFT;
        break;
    case 0b00001000:
        btns |= HAT_RIGHT;
        break;
     case 0b00000101:
        btns |= HAT_UP_LEFT;
        break;
     case 0b00001001:
        btns |= HAT_UP_RIGHT;
        break;
     case 0b00000110:
        btns |= HAT_DOWN_LEFT;
        break;
    case 0b00001010:
        btns |= HAT_DOWN_RIGHT;
        break;    
    default:
        btns |= HAT_CENTER;
  };

  //перемещение кнопок: 16->20, 17->16, 18->17, 19->18, 20->19
  bool tmp=bitRead(btns,23);
  bitWrite(btns, 23, bitRead(btns, 19));
  bitWrite(btns, 19, bitRead(btns, 20));
  bitWrite(btns, 20, bitRead(btns, 21));
  bitWrite(btns, 21, bitRead(btns, 22));
  bitWrite(btns, 22, tmp);
  
  buttons=btns;
}