vsulako/AFFBWheel #45: Поиск центра без стопорных ограничений

Envi4, 2024-01-09:
Здравствуйте, а есть ли возможность сделать центровку при подключении к ПК, без стопорных механизмов, например торможением двигателя, а центр как то сохранять в прошивку значением энкодеры, а не силен в программировании, но если такая возможность есть, очень был бы рад ее увидеть в будущем

vsulako, 2024-01-09:
При включении и так текущее положение руля считается центром. 
Просто выровняйте его перед подключением, или нажимайте кнопку центровки.
А если речь о том чтобы руль сам крутился и находил центр, то без механических ограничителей непонятно как он должен понять, где находится центр.  Разве что по Z-сигналу от энкодера, но он есть не на всех энкодерах. На моем нету.

lazenyuk-dmitry, 2024-01-09:
> При включении и так текущее положение руля считается центром. Просто выровняйте его перед подключением, или нажимайте кнопку центровки. А если речь о том чтобы руль сам крутился и находил центр, то [...]

Вроде все так. Только по Z сигналу. Если нету ограничителей. Иначе никак... И в прошивку сохранять это все Z по сигналу. Ну была бы прикольная фича) Может кто то тебе задонатит на энкодер с Z сигналом))

@Envi4, а вы заметили, что в описании есть про кнопку для центровки? Вы можете назначить любую кнопку и центрировать руль. Энкодеры с Z сигналом подороже будут.

closed 2024-01-14 by Envi4