vsulako/AFFBWheel #47: Проблемы с центрированием

Envi4, 2024-01-14:
Приветствую, решил сделать стопорные кольца для поиска центра, сделал модель, распечатал, поставил, но столкнулся с проблемой, подключаю руль, он начинает искать центр вначале влево, после направо, а после снова до упора влево и не останавливает двигатель упираясь в стопор, так же при подключении руля вместе с двигателем после этих махинаций поиска центра, он не видится системой, пишет no connection, подключается только если отключить двигатель и подключить руль без него

Darkmen203, 2024-01-18:
> Приветствую, решил сделать стопорные кольца для поиска центра, сделал модель, распечатал, поставил, но столкнулся с проблемой, подключаю руль, он начинает искать центр вначале влево, после направо, [...]

а какая у вас скорость и при этом все правильно отображается в программе драйвера?

Envi4, 2024-01-18:
К сожелению с ком порта не получилось взять данные, игрался с разными параметрами и в итоге сломал стопорные кольца и пришлось отключать плату, буду переделывать из опытом скажу, несколько раз у меня получалось завершить настройку и центр находило, каждый раз повышая силу мотора, с force 9500 начал более менее работать, но с такими настройками становится страшно использовать эту функцию, так при таких значениях он не факт что найдет, а 10000 сломало ограничитель

Darkmen203, 2024-01-18:
А что у вас за мотор?) У меня проблема ровно наоборот, я рукой прокручиваю мотор, потому что он импульсами сопротивляется

Envi4, 2024-01-18:
> А что у вас за мотор?) У меня проблема ровно наоборот, я рукой прокручиваю мотор, потому что он импульсами сопротивляется

pik 8-6/2.5 работает от 24v

Darkmen203, 2024-01-18:
Хм, ну это довольно мощный мотор, с которым обычно не используют ограничители, он сам может выступать ограничителем

Envi4, 2024-01-18:
> Хм, ну это довольно мощный мотор, с которым обычно не ограничивают ограничители, он сам может поддерживать ограничитель

Мне ограничитель нужен для создания автоматической центровки, дальше он не будет использоваться, раньше использовал по кнопке, но автор добавил такую возможность и я загорелся создать такую центровку, но пока есть проблемы

Darkmen203, 2024-01-18:
А вы на вал ставили руль уже? Вполне возможно, что вам силу приходится так увеличивать из-за того что руль тяжёлый и двигателю нужно много усилия прилагать чтобы прокрутить и он поэтому ваши стопоры ломает при этом не завершая процедуру калибровки

Envi4, 2024-01-18:
> А вы на вал уже поставили руль? Вполне возможно, что вам придется столкнуться с такой необходимостью из-за того, что руль тяжёлый и двигателю нужно приложить много усилий, чтобы прокрутить, и поэтому ваши стопоры ломаются при этом не завершая процедуру процедуры.

Без руля тестировал, он съёмный у меня, поэтому не понимаю почему нужна такая силу ему, спокойно одним пальцем прокручивается
А от стопорных колец сейчас отказался после их поломки пришла идея сделать резьбовой ограничитель

Darkmen203, 2024-01-18:
Хотел тоже вам сказать что можно попробовать гайкой, получается у вас на границах стоит по две гайки в виде контргаек и где-то по середине  ещё одна гайка, которая по-идее будет от ваших движений закручиваться или откручиваться, что-то подобное у меня получалось в первой конструкции, но были минусы, что чутка закусывали вал из-за этого, т.к. он у меня упирался в уголок (да и цель гайки была не быть ограничителем, а просто в виде контргайке держаться, но я забыл затянуть, хех)

jhnf, 2024-01-20:
> К сожалению, предположений нет. Двигатель, судя по всему, вращает в правильную сторону (если влево это против часовой) То что во время поиска центра контроллер недоступен - это так и должно быть, так было проще сделать. Немного обновил код, посмотрите что при поиске центра выводит в ком-порт. (по команде "autocenter")

А почему не сделать нахождение крайних положений через концевые выключатели, типа тех, что в 3Д принтерах работают? Кольцевой ограничитель упирается в выключатель, тот срабатывает, мотор стопорится, кольца целые. И повесить их не на виртуальные кнопки, а на пару свободных портов микроконтроллера.

vsulako, 2024-01-21:
> А почему не сделать нахождение крайних положений через концевые выключатели, типа тех, что в 3Д принтерах работают? Кольцевой ограничитель упирается в выключатель, тот срабатывает, мотор стопорится, кольца целые. И повесить их не на виртуальные кнопки, а на пару свободных портов микроконтроллера.

Можно и так. Надо будет как нибудь добавить.