vsulako/AFFBWheel #52: Проблема с использованием MCP23017

mitiaborodada, 2024-02-04:
Привет!
Очень нужна помощь с прооблемой использования модуля MCP23017.
Всё сделал согласно схеме, но почему-то, при подключении к ПК сразу срабатывают 01 / 09 кнопки, иногда 01/ 02 /09 /10
[image]

mitiaborodada, 2024-02-04:
Дополнение. 
Видимо был где-то плохой контакт, т.к. сейчас при включении срабатывают сразу 6 кнопок: 01, 02, 03  и 09, 10, 11 (физически кнопки даже не подключены к платам)
Причём, остальные кнопки работаю нормально, если каждую замкнуть на МИНУС.

vsulako, 2024-02-04:
Попробуйте увеличить значение параметра I2C_DELAY.

mitiaborodada, 2024-02-05:
Не работает. Попробовал поменять пины:
#define I2C_PIN_SDA 2
#define I2C_PIN_SCL 3
и I2C_DELAY делал от1 до 5

При подключении параллельно AS5600 и MCP23017 вообще ничего не работает, если оставить AS5600, то датчик поворота работает, Но если вообще отключить датчики от пинов 2 и 3, то  срабатывают все кнопки сразу. А датчик MCP23017 отдельно никак не хочет работать. Даже не знаю куда копать.

mitiaborodada, 2024-02-10:
> Попробуйте увеличить значение параметра I2C_DELAY.

Приветствую!

Уже проверил несколько раз схему и все соединения прозвонил и перепаял. Сначала думал, что это из-за того, что модули разнесены - один стоит в КПП, другой на руле.  Но потом модули поставил рядом. Провода, соединенияющие модули по  SDA и SCL минимальны - 5 см. 

Пробовал 2C_DELAY с разными значениями от 1 до 400. Руль при этом работает, а кнопки не хотят. Т.е. работают, кроме тех, что на скрине - они постоянно включены. При некотрых значениях 2C_DELAY добавляется ещё по одной включенной кнопки с каждой стороны. 

Вообще уже не понимаю в чём косяк.

Буду очень рад, если поможете решить проблему с модулями.

mitiaborodada, 2024-02-11:
В дополнение.  Если поменять адреса модулей, например так:
MCP23017_ADDR1  0x22
MCP23017_ADDR2  0x24
то тогда вообще никакой реакции нет. Ни одна кнопка не включается. Разные варианты пробовал, ничего не помогает.

jhnf, 2024-02-14:
@mitiaborodada а длина проводов какая?

вот что нашёл:
> при разработке устройств с MCP23017 (и всех аналогичных по принципу работы) следует избегать подключение к шине I2C датчиков и сенсоров с длинными проводами, по возможности их стоит перенести на другую шину (например на ESP32 их две)

mitiaborodada, 2024-02-14:
> @mitiaborodada а длина проводов какая?
> 
> вот что нашёл:
> 
> > при разработке устройств с MCP23017 (и всех аналогичных по принципу работы) следует избегать подключение к шине I2C датчиков и сенсоров с длинными проводами, по возможности их стоит перенести на другую шину (например на ESP32 их две)

5 см. Я как раз это написал во втором сообщении. До этого модули были разнесены, но результат тот же. 

Между модулями и контролёром 5 см.

Вчера пробовал вообще отключить всё, кроме этих модулей. Пробовал разные адреса для них, но единственное чего добился, это одна включенная кнопка постоянно. При включении остальных кнопок могут сработать сразу две в разных местах. Думаю, может быть, проблема с модулями? Может брак попался?

mitiaborodada, 2024-02-19:
Надоели танцы с бубном с этими модулями. Решил пойти самым простым путём. Большинство кнопок разделю на блоки по 6-7 штук и повешу на разные аналоговые входы. Т.к. большинство кнопок работают сами по себе и не требуют проверки одновременного нажатия, то это самый быстрый и простой вариант, не требующий глобальных изменений в коде прошивки. 
Исключением являются  2 кнопки на ручке КПП, отвечающие за переключение рядов и уровень передач.