vsulako/AFFBWheel #53: Проблема с настройкой педалей

IvanZagrebin, 2024-02-10:
В утилите аффбвхил ось встает в упор при нажатой педали на половину(кнопка Макс не помогает)

mitiaborodada, 2024-02-11:
> В утилите аффбвхил ось встает в упор при нажатой педали на половину(кнопка Макс не помогает)

1. Какие датчики используете на педалях?
2. На всех педалях такая проблема?
3. Какую схему используете для педалей? Их автор предлапгает несколько.
4. При нажатии на педаль данные, которые считываются с датчиков, плавно нарастают?

PS Это как в службе поддержке, по тексту клиента ничего не понятно, кроме того, что не работаетЮ поэтому столько вопросов)))

IvanZagrebin, 2024-02-11:
Использую потенциометры, на всех педалях такая проблема, педаль достаточно плавно.
[image]

IvanZagrebin, 2024-02-11:
Может лучше в ВК перейти

IvanZagrebin, 2024-02-11:
Так удобнее

lazenyuk-dmitry, 2024-02-11:
> > В утилите аффбвхил ось встает в упор при нажатой педали на половину(кнопка Макс не помогает)
> 
> 1. Какие датчики используете на педалях?
> 2. На всех педалях такая проблема?
> 3. Какую схему используете для педалей? Их автор предлапгает несколько.
> 4. При нажатии на педаль данные, которые считываются с датчиков, плавно нарастают?
> 
> PS Это как в службе поддержке, по тексту клиента ничего не понятно, кроме того, что не работает, поэтому столько вопросов)))

Я общался с Иваном ВК) Вот чето я думаю что косяк именно на стороне оборудования. Похожих проблем ни у кого не было. У меня педали на простых потенциометрах. Все четко. Так же есть стик и одна вертелка и тоже все отлично работает.

@IvanZagrebin а потенциометры новые? Какие. Скинь сюда фотки своих печалей. Потенциометры и как подключено.

IvanZagrebin, 2024-02-11:
Потенциометры из джойстика старого, но работал он нормально

lazenyuk-dmitry, 2024-02-11:
> Потенциометры из джойстика старого, но работал он нормально

Опа... А вот и проблема наверно) я ставлю все на косяковые потенциометры. Тем более из джойстика. Даже не знаю как ты их туда вкорячил. Я знаю в стихах стоят они мелкие капец. Поставь нормальные потенциометры. Хоть из старой советской техники.

IvanZagrebin, 2024-02-11:
Да они норм, люфта нет вроде

IvanZagrebin, 2024-02-11:
Можно конечно на мультиметре проверить

IvanZagrebin, 2024-02-11:
Крепил на суперклей

IvanZagrebin, 2024-02-11:
Кстати так подумал может затек

IvanZagrebin, 2024-02-11:
Я прошивку перезаливал

IvanZagrebin, 2024-02-11:
Попробую

lazenyuk-dmitry, 2024-02-11:
Ну проверь сперва потенциометры. Прошить успеешь.

IvanZagrebin, 2024-02-11:
потенциометр подозрительно работает

IvanZagrebin, 2024-02-11:
показание до 20 ком а потом 0

IvanZagrebin, 2024-02-11:
сейчас перепаяю
на новый

IvanZagrebin, 2024-02-11:
видимо дело в потах, с новым ситуация получше(из за потенциометра ось скачет, но ему тоже плохо
 )

IvanZagrebin, 2024-02-11:
ура! проблема решена

IvanZagrebin, 2024-02-11:
спасибо за помощ