vsulako/AFFBWheel #54: Подскажите как подключить gimx

Envi4, 2024-02-15:
Прикупил cp2102, не могу понять как правильно ли я понимаю как сделать или нет
Соединить TXD, RXD, GND с моей платой ардуино леонардо и подключить gimx и леонардо по юсб в 2 порта? И так же если этот вариант правильный, можно ли cp2102 повесить на тот же usb выход что и ардуино леонардо 
Или надо еще отдельную плату ардуино иметь к cp2102?

lazenyuk-dmitry, 2024-02-15:
Собираешь GIMX. Если играешь на ПК то нужно включить оба usb в ПК. Потом подключить руль к ПК и настроить кнопки руля на эмуляцию в коныигураторе GIMX. У тебя будет и твой пуль в диспетчере устройств и эмулятор руля например g25.

Envi4, 2024-02-18:
Благодарю все получилось, но столкнулся с такой проблемой что Форза хорайзен все адекватно работает, а решил попробовать мотоспорт новый и gimx просто крашется с ошибкой невозможно обновить эффект, может знаете как можно решить

closed 2024-06-18 by Envi4