vsulako/AFFBWheel #57: Use the digital pin to map action

5nax, 2024-03-10:
I am very new in programming this sort of stuff so please help me through. I have a couple of digital pins free on my pro micro. is it possible to map the free digital pins ( not used for anything else ) to be mapped to the HAT buttons? for example if I have pin number 8 free 
`#define HAT_BTN_DOWN   8
`
will the hat button down work? 

```
#define HATSWITCH           //uncomment to enable feature
#define HAT_BTN_UP     20   //button numbers for hat directions
#define HAT_BTN_DOWN   21
#define HAT_BTN_LEFT   22
#define HAT_BTN_RIGHT  23
#define HAT_CLR_BTNS   	    //if this line is commented, selected buttons will continue to register presses along with hat switch

```

vsulako, 2024-03-11:
Hatswitch config does not use pin numbers, it uses controller button numbers. 

If you want to map arduino pin as button, use direct-pin-to-button section.

Example: 
```
//buttons directly connected to pins 
#define DPB        //Enable
#define DPB_PINS       3,4,5,8  //comma-separated list of pins
#define DPB_1ST_BTN    1
```
Here, pins 3,4,5,8 will be mapped to buttons 1,2,3,4.  These buttons should connect selected pin to GND when pressed.

Then, you can select these buttons (1,2,3,4) as hat switch directions.

5nax, 2024-03-11:
Thanks for the input, I got it working.

But now i have another issue i am trying to use an analog joystick module for H Shifter

```
//analog H-shifter
#define ASHIFTER
#define ASHIFTER_PINX     A7
#define ASHIFTER_PINY     A8
#define ASHIFTER_POS      8   //6 or 8 positions
#define ASHIFTER_Y1       137 
#define ASHIFTER_Y2       255
#define ASHIFTER_X1       0  
#define ASHIFTER_X2       139  
#define ASHIFTER_X3       199  
#define ASHIFTER_1ST_BTN  25
```

Is it possible to use average value for the bounding zones for x1,x2,x3 and y1,y2 cause my joystick sends unstable value 2-3 offset.

5nax, 2024-03-11:
Could i use the bottom code to add tolerance?

```
//analog shifter
#ifdef ASHIFTER
  // Assume TOLERANCE is defined somewhere in your code, e.g., #define TOLERANCE 10
  uint8_t x = analogReadFast(ASHIFTER_PINX) >> 2;
  uint8_t y = analogReadFast(ASHIFTER_PINY) >> 2;
  uint8_t g = 0;

  if (ashifter_out) {
    Serial.print(F("Analog shifter X:"));
    Serial.print(x);
    Serial.print(" Y:");
    Serial.println(y);
  }
  
  // Clear bits for different positions
  #if ASHIFTER_POS == 8
  wheel.buttons &= ~((uint32_t)0xff << (ASHIFTER_1ST_BTN - 1));
  #elif ASHIFTER_POS == 6
  wheel.buttons &= ~((uint32_t)0x3f << (ASHIFTER_1ST_BTN - 1));
  #endif

  // Set bits with tolerance
  if (y <= ASHIFTER_Y1 + TOLERANCE && y >= ASHIFTER_Y1 - TOLERANCE) {
    g = 0b00000001;
  } else if (y >= ASHIFTER_Y2 - TOLERANCE && y <= ASHIFTER_Y2 + TOLERANCE) {
    g = 0b00000010;
  } else {
    return; // If Y is not within tolerance of Y1 or Y2, exit early
  }

  if (x >= ASHIFTER_X1 - TOLERANCE && x < ASHIFTER_X2 - TOLERANCE) {
    // No need to shift g, as it's already set for the first range
  } else if (x >= ASHIFTER_X2 - TOLERANCE && x < ASHIFTER_X3 - TOLERANCE) {
    g <<= 2; // Shift for the second range
  } 
  #if ASHIFTER_POS == 8
  else if (x >= ASHIFTER_X3 - TOLERANCE) {
    g <<= 2; 
  }
  if (x >= ASHIFTER_X3 - TOLERANCE && x < ASHIFTER_X4 - TOLERANCE) {
    g <<= 2; 
  }
  else if (x >= ASHIFTER_X4 - TOLERANCE) {
    g <<= 4; 
  } 
  #endif
  
  wheel.buttons |= ((uint32_t)g << (ASHIFTER_1ST_BTN - 1));
#endif
```

i used the help of chatgpt for this code and the gear shifter is working as expected but i dont know if it affects any other codes..

vsulako, 2024-03-12:
Yes, it should not affect anything.

5nax, 2024-03-13:
Thank you! Appreciate it.

closed 2024-03-13 by 5nax