vsulako/AFFBWheel #58: Auto find range problem

Ertugrulbabaci, 2024-03-25:
Hello. I added physical limiters in two directions, left and right. I set it to 900 degrees. When I activate it as you say in the codes, it goes first to the left and then to the right. Then, instead of finding the middle point, it starts to turn left forever. There is no midpoint. what is the reason of this?

vsulako, 2024-03-25:
I changed code a bit, please update and check if it helps.

Ertugrulbabaci, 2024-03-26:
It now finds the midpoint, but there is a left turn of -8 degrees. Can I make it stop at the midpoint? Also, can a button be assigned to operate this system?

Ertugrulbabaci, 2024-03-26:
It now finds the midpoint, but there is a left turn of -8 degrees. Can I make it stop at the midpoint? Also, can a button be assigned to operate this system? https://github.com/vsulako/AFFBWheel/assets/154468129/c8e1324c-6a91-43bf-8909-780a86037604

oscarMO7, 2024-03-29:
I added this feature it is really useful but it would be better if I you assign it to a button @vsulako

vsulako, 2024-04-03:
> It now finds the midpoint, but there is a left turn of -8 degrees. Can I make it stop at the midpoint? 
try to lower AFC_FORCE. Applied force turns of when reaching center, but wheel can go further because of inertia. 
Need to rewrite this procedure.

> Also, can a button be assigned to operate this system? 
Added this.  `#define AFC_BUTTON` in config.h