vsulako/AFFBWheel #59: Пропали звуки тормозной системы в ATS и ETS

mitiaborodada, 2024-04-02:
Привет!
У меня возникла такая проблема. В какой-то момент в ATS пропал звук тормозов (когда нажимаешь на педаль слышны всякие "пшыкающие" звуки)
Пропал именно звук тормоза. Все остальное работает. Что интересно, такая же проблема и с ETS 2. 
Заметил одну деталь, если вместо педали использовать клавиатуру, то звук тормозов есть. Как такое может быть??? Что произошло в системе, что реакция на педаль перестала работать? 

Может есть какие-то идеи? Очень нужна помощь. Буду очень благодарен 🙏

vsulako, 2024-04-03:
Чертовщина какая то. Даже не представляю, в чем может быть дело.

mitiaborodada, 2024-04-04:
Оно как бы не критично, но эффект симуляции пропадает😁 И уже просто становится интересным, в чем причина. Сначала, переустановил сам мод, отвечающий за звуки. Не сработало, вспомнил, что у меня есть же ETS, захожу туда, а там тоже самое, звук тормозов пропал. Значит дело не в игре. Возможно ошибка кроется в usb - драйвер, либо виртуальный com порт. У меня ещё на ноуте установлена игра, там попробую.

mitiaborodada, 2024-04-04:
Как говорится, дело было не в бобине))) 
Попробовал на нотнике. Всё работает.
Значит проблема в ПК. Но как и почему??? Это точно не с игрой проблема, т.к. звук сразу пропал в обеих играх одновременно. Значит проблема в передаче данных через протоколы обмена.