vsulako/AFFBWheel #61: 2 joy 16 b

Brandy0841, 2024-05-28:
Nice work, congratulations. I have a question. Is it possible to make a steering wheel with 2 joysticks and 16 buttons? It would be nice to be able to use 2 joysticks for games like farming simulator and snowrunner. Thanks! @vsulako

vsulako, 2024-06-16:
I don't understand what do you mean by 2 joysticks. Are they thumbsticks? 
Gaming controller can have up to 8 analog axes. 1 axis is wheel, 3 are for pedals, and there are 4 spare axes, so nothing prevents from connecting 2 thumbsticks (2 axes each).

Ertugrulbabaci, 2024-06-21:
Is the maximum limit that Arduino can take 8 axes or is it related to the codes written? For example, can't you change the codes to 10 axes?

Ertugrulbabaci, 2024-06-21:
> I don't understand what do you mean by 2 joysticks. Are they thumbsticks? Gaming controller can have up to 8 analog axes. 1 axis is wheel, 3 are for pedals, and there are 4 spare axes, so nothing prevents from connecting 2 thumbsticks (2 axes each).

Is the maximum limit that Arduino can take 8 axes or is it related to the codes written? For example, can't you change the codes to 10 axes?

vsulako, 2024-06-24:
It is DirectInput limitation. Windows will only recognize up to 8 axes and 128 buttons in a single HID gaming controller, even if more are declared, excess ones will not work.
Also, there are only 8 axis names in HID specifications for a generic controller: x,y,z, rx,ry,rz, slider,dial. May be it matters too.