vsulako/AFFBWheel #62: 2 ProMicros: Steering Wheel + Pedals and Buttonbox (74HC165) + shifter (74HC165 using limit switch)

spry02, 2024-06-15:
Hi, first of all I want to make two separate circuits due to analog limitations (can be changed to one circuit but don't know if I can use enough analog inputs, one of the joysticks can be removed)
### **First circuit:**
I want to make few buttons, two joysticks, steering wheel and pedals
### **Second circuit:** 
7 buttons gearbox with buttonbox + handbrake as analog 

I wonder if you have any idea about functionality of this setup, I want to use this steering wheel with ETS2.

**My question/problem:**
I want to make gearbox with two selectable ranges using bistable switch. I thought about solution like this (Using 7 limit switches, two 74HC165 with separated DATA pins selectable with switch. Limit switches will be connected to both HC165, both HC circuits are linked together, but as I told DATA connection will be selected with bistable switch.) 

[image]

### Would it work as for example: 
Range 1 buttons: 1, 2, 3, 4, 5, 6, 7
**and then after switch flick**
Range 2 buttons: 8, 9, 10, 11, 12, 13, 14
(in my mind it works like: you engage first gear, then flick the switch, and now it is seventh gear)

**or**

buttons will override as same inputs? 
(you engage first gear, then flick the switch, and it's still first gear)

If you have any advices or better idea to fulfill my idea please let me know. I appreciate every form of help.

vsulako, 2024-06-16:
This circuit will work on second scenario (engage first gear, then flick the switch, and it's still first gear).
Switch does nothing here, both sides of switch leads to identical circuitry.
You don't need second HC165, just make switch change DATA wire from arduino's side (in one position DATA from HC165 connects to pin 2, in other position it connects to pin 7).
On the other side,  is this really necessary? ETS2 supports Range/Splitter buttons, which should do the same thing: when Range button is unpressed, shifter changes 1,2,3,4,5,6 gears, when it is pressed - they becomes 7,8,9,10,11,12 gears. 
https://steamcommunity.com/sharedfiles/filedetails/?l=polish&id=1215476496
Also, ETS2 can only assign 6 keys for forward gears.

spry02, 2024-06-17:
Damn thanks a lot, i honestly didn't look to much into ets2 settings, kinda delayed that to other time because I'm still completing my parts and gaining enough budget, thanks to you I won't make such a stupid mistake. So now I can really easily make 24 buttons on my button box and still have fully functional gearbox for trucking with friends. 

When I finally begin to make whole setup I will keep posting updates of my work if you are interested. Also when I won't be sure of some solutions I will ask for help. 

Once again, thanks a lot mate. You are making really great job! Take care!