vsulako/AFFBWheel #66: Vibration at center

cartaio, 2025-02-06:
Hi, i've been using your code for more than a year, i'm the guy from issue 35... work flowlesly other that i have in some games (assetto corsa is the worst for now), where i have a lot of vibration at the middle... one issue i found is that if i do the ac lut calibration it wont complete because it cannot find the center for some reason, do anyone had the same problem? i'll put a video if needed to explain it better...
thank you.

vsulako, 2025-02-20:
What is ac lut calibration?

cartaio, 2025-02-21:
It's really bad because it cannot get to the center when you try to calibrate it... i will upload a private video on yt so i can show you what it does...

cartaio, 2025-02-25:
[https://www.youtube.com/shorts/lJTymzJuInE?feature=share](url) Video of lut generation(yes is this bad in game too)
[https://imgur.com/a/jAePHpk](url) lut results...