vsulako/AFFBWheel #68: Инвертирован FFB в левую сторону для игры Forza Motosport

swellskam, 2025-02-11:


swellskam, 2025-02-11:
Такая же проблема наблюдается и с прошивкой VNWheel  от туда Вы брали HID. Возможно где то HID описание не правильно задано. Я не силен в этом, но я тоже программирую на языке C# хотелось бы понять где копать эту проблему.

vsulako, 2025-02-20:
Судя по тому что удалось найти в интернете, эта проблема в Forza Motorsport встречается и на других устройствах, .
Как будто бы игра игнорирует знак обратной связи, должно быть в отрицательную сторону, а она дает команду на положительную. Инвертирование ОС в игре у меня вообще не работало.
Если на той прошивке работает правильно, то можно попробовать разобраться. Попробую посмотреть, если будет время, но ничего обещать не могу.

vsulako, 2025-02-22:
Кажется, нашел причину.
Надо проверить.
Вот файл. [FfbEngine.zip](https://github.com/user-attachments/files/18924317/FfbEngine.zip)
Внутри ffbengine.cpp, замените им тот, что в папке проекта, прошейте заново и попробуйте снова.

swellskam, 2025-02-22:
Спасибо что ответили да в этом файле я его уже исправил там же и обратной связи не было на Gimx, но для себя я переделал его кардинально я работал над обратной связью в целом, а так же сделал регулировку в процентах. Спасибо еще раз очень хорошая работа фитбек отличный.

swellskam, 2025-02-22:
https://prnt.sc/G-d3s4YCDDoV

swellskam, 2025-02-22:
Мое решение вот такое https://prnt.sc/juWcoGJGw02H

vsulako, 2025-02-22:
Вообще впервые вижу чтобы игра использовала параметр directionX. 
Поэтому черт его знает, как правильно. 

> https://prnt.sc/juWcoGJGw02H
По моему, на скрине потенциальный баг - меняется знак не конкретного эффекта, а totalForce.  
А если одновременно создать два эффекта с directionX>127? знак поменяется дважды.  
Наверное, вряд ли такое произойдет, но вдруг?

Ну и кстати еще можно оптимизировать немного. зачем каждый раз рассчитывать знак? я бы перенес это в обработку SetEffect, а в цикле ОС один раз умножал на заранее сохраненный знак в int8_t.

swellskam, 2025-02-22:
Ну я делал на ощуп а не по научному. Дело в том что я со сдвигами не дружу я понимаю что на сдвигах быстрее работает но пока я до них не дорос я не силён C++и потому я понимаю как есть но если это работает правильно на ощуп значит мне кажется то и пусть работает. Кстати применил я ваше исправление правда уже в свем коде пропала сила раза в три а то и четыре.

vsulako, 2025-02-22:
Может, вы переделывали hid descriptor и изменяли пределы magnitude у Constant effect?
Ну или мой синус неверно работает.

swellskam, 2025-02-22:
Нет я не трогал дескриптор там все так и осталось? но я убрал сдвиги в aplyGain изменил возвращаемый тип для всех эффектов он сейчас у меня int32_t потому я применил totalForce += к каждому эффекту убрал tmpForce в общем сделал так как я встречал в других кодах. Я просто ну не понимаю я сдвигов. И все это я делал с помощью deepseek это типа gpt chat. Да были проблемы с переполнениями но с горем пополам я это все осилил. Я поубирал бесполезные мне прибомбасы сделал для своего руля g29 он у меня сгорел и я решил сделать на Arduino сейчас фитбек на руле лучше чем был он с завода а самое главное множество настроек. Основная работа Ваша я не спорю но там не нужного было много для меня. Тем самым я сделал передачу в конфигурационную программу я сделал json так как места много освободилось. Программа еще не доделана но скриншоты я приложу Вы сами поймете что лучше обратить внимание на то что нужнее. https://prnt.sc/kWIztVMY7Mul      https://prnt.sc/W6G-QqAHAXh_

swellskam, 2025-02-23:
Да кстати я проверил Ваше исправление в Вашей прошивке да на вашем не тронутом коде исправление актуально только минус спереди не нужен он инвертирует то есть тянет от средины до края хотя в EndStop упирается корректно. Спасибо, что не игнорировали меня.

ertoben, 2025-07-26:
Can you share the version of the file that works without any problems in Forza Motosport? @swellskam @vsulako